Xilinx/Vitis-AI #2: Vitis-AI can support aws f1 ？

zyxcambridge, 2019-12-03:
update the ml-suite and  give a ami

wilderfield, 2019-12-03:
Stay tuned. We will build an AMI shortly.

zyxcambridge, 2019-12-09:
XILINX_XRT      : /opt/xilinx/xrt
PATH            : /opt/xilinx/xrt/bin:/root/miniconda3/bin:/root/miniconda3/condabin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games
LD_LIBRARY_PATH : /opt/xilinx/xrt/lib:
PYTHONPATH     : /opt/xilinx/xrt/python:
WARNING: The xbutil sub-command flash has been deprecated. Please use the xbmgmt utility with flash sub-command for equivalent functionality.

Card not found!
WARNING: The xbutil sub-command flash has been deprecated. Please use the xbmgmt utility with flash sub-command for equivalent functionality.

zyxcambridge, 2019-12-09:
Verifying XILINX_XRT
---------------------
XILINX_XRT      : /opt/xilinx/xrt
PATH            : /opt/xilinx/xrt/bin:/root/miniconda3/bin:/root/miniconda3/condabin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games
LD_LIBRARY_PATH : /opt/xilinx/xrt/lib:
PYTHONPATH     : /opt/xilinx/xrt/python:
--- Using System XCLBIN ---
Couldn't find the VAI compiler. Exiting ...

aws f1 ubuntu 16.04

imyoungyang, 2020-05-21:
Any update on this ticket?

Looks the XRT runtime tool xbutil, and xbmgmt can't communicate to AWS F1 card.

Second, the demo code only provide ZCU102, ZCU104, U50 boards. Could you provide VU9P, which is  Virtex UltraScale+ AWS VU9P F1  Acceleration Development Board?

rodrigomelo9, 2020-07-18:
No news? The AMI is not strictly needed but the VU9P support? The 1.2 version was recently launched without this addition :-(

wilderfield, 2020-07-18:
We are looking at enabling AWS EC2 F1 by Vitis AI 1.3

Bryan

rodrigomelo9, 2020-07-18:
The idea is to support the VU9P or only to provide an AMI to replace the ML-suite?

acastanedam, 2021-02-12:
Hi,

Are there still plans for supportting AWS's FPGAs?. Maybe there is something I am overlooking, but I can not use it for VU9Ps, even with the most recent version. It would be great to use Vitis-AI in AWS.

Regards,

A. Castaneda

HasinduKariyawasam, 2022-01-04:
After setting up Vitis AI on AWS f1.2xlarge instance, I tried to run an [example](https://github.com/Xilinx/Vitis-AI/tree/master/examples/DPUCADF8H/tf_resnet). In that, there are steps for [setting up Alveo accelerator cards](https://github.com/Xilinx/Vitis-AI/tree/master/setup/alveo#alveo-card-setup-in-host). But the AWS f1.2xlarge instance has an Ultrascale+VU9P card. Do I need to set up that? If so are there any tutorials or guides on how to do that?

fanz-xlnx, 2022-09-01:
To start from F1 instance, you don't need to worry about the shell or xclbin, but need to install the XRT and XRM yourself. The version on AWS is different from the release version, which you may refer to the guidence [here](https://github.com/Xilinx/Vitis-AI/tree/master/docs/aws).

WWWindrunner, 2022-12-09:
> 

Does Xilinx support aliyun f3? eg. https://www.aliyun.com/product/ecs/fpga.
I saw that there are versions for aws and azure in the document, but not for aliyun.
For cloud users, do you need to set xclbin? If you don’t set it, you will get an error, but the setting has developed aws-xclbin specifically for aws.

WWWindrunner, 2022-12-09:
> To start from F1 instance, you don't need to worry about the shell or xclbin, but need to install the XRT and XRM yourself. The version on AWS is different from the release version, which you may refer to the guidence [here](https://github.com/Xilinx/Vitis-AI/tree/master/docs/aws).

WWWindrunner, 2022-12-09:
> To start from F1 instance, you don't need to worry about the shell or xclbin, but need to install the XRT and XRM yourself. The version on AWS is different from the release version, which you may refer to the guidence [here](https://github.com/Xilinx/Vitis-AI/tree/master/docs/aws).

Does Xilinx support aliyun f3? eg. https://www.aliyun.com/product/ecs/fpga.
I saw that there are versions for aws and azure in the document, but not for aliyun.
For cloud users, do you need to set xclbin? If you don’t set it, you will get an error, but the setting has developed aws-xclbin specifically for aws.

WWWindrunner, 2022-12-09:
> To start from F1 instance, you don't need to worry about the shell or xclbin, but need to install the XRT and XRM yourself. The version on AWS is different from the release version, which you may refer to the guidence [here](https://github.com/Xilinx/Vitis-AI/tree/master/docs/aws).
For cloud users, do I need to set xclbin? If I don’t set it, an error will be reported. There is aws-xclbin for aws, but I don't see aliyun-xclbin. 
eg.  https://www.aliyun.com/product/ecs/fpga

WWWindrunner, 2022-12-09:
> To start from F1 instance, you don't need to worry about the shell or xclbin, but need to install the XRT and XRM yourself. The version on AWS is different from the release version, which you may refer to the guidence [here](https://github.com/Xilinx/Vitis-AI/tree/master/docs/aws).

In aws, looking at the document shows that xbutil can find the platform of aws, but on the fpga cloud server of aliyun, xbutil cannot find the corresponding platform, and the error message shows that the device cannot be found. Why?