Xilinx/Vitis-AI #5: GPU Container link

SpontaneousDuck, 2019-12-05:
Downloading the GPU tools container gives a bad and very small download. When I navigate in a browser to the link I get a 404 message from the Xilinx website.

wilderfield, 2019-12-05:
Thanks @SpontaneousDuck . We decided not to distribute a prebuilt Docker Image for GPU accelerated quantization. We will distribute a Dockerfile to allow users to build it themselves. Please give us a few days to resolve this. I will update the instructions now. For now, just use the CPU tools container. It provides the same tools, just not GPU accelerated.

shua1zhang, 2019-12-22:
Hi @SpontaneousDuck ,

Now the GPU docker building scripts are released in https://github.com/Xilinx/Vitis-AI/tree/master/docker. Please check.

Thank you.

closed 2019-12-22 by shua1zhang

xiaoxiaoxiaoxiaotao, 2021-08-09:
> Thanks @SpontaneousDuck . We decided not to distribute a prebuilt Docker Image for GPU accelerated quantization. We will distribute a Dockerfile to allow users to build it themselves. Please give us [...]

Hello, I would like to ask that my network environment is not good, and I always fail to use docker_build_gpu.sh. Could you provide the compiled gpu_docker environment? Thank you!

hanxue, 2021-08-09:
Hi @xiaoxiaoxiaoxiaotao , we notice that some Ubuntu package sources fail to work from within China. One alternative is to use China sources, for example https://github.com/Xilinx/Vitis-AI/blob/41a397a08139c1a8ddeceedda70a83f33d4f4d4f/docs/quick-start/install/install_docker/README.md

Do you mind pasting the docker_build_gpu.sh error that you face?

xiaoxiaoxiaoxiaotao, 2021-08-10:
> Hi @hanxue . The error is 
E: The repository 'https://developer.download.nvidia.com/compute/cuda/repos/ubuntu1804/x86_64  Release' does not have a Release file.
E: The repository 'https://developer.download.nvidia.com/compute/machine-learning/repos/ubuntu1804/x86_64  Release' does not have a Release file.

And I can't this website.

hanxue, 2021-08-10:
Hi @xiaoxiaoxiaoxiaotao , thanks for sharing the error. Let's try to replace the US/international nvidia Ubuntu apt sources with alternative sources in China. 

Remove (backup) the nvidia CUDA and machine-learning repos:

```
sudo mv /etc/apt/sources.list.d/cuda.list /etc/apt/sources.list.d/cuda.list.BACKUP
sudo mv /etc/apt/sources.list.d/nvidia-ml.list /etc/apt/sources.list.d/nvidia-ml.list.BACKUP
```

Add the Aliyun mirror for CUDA

```
curl -fsSL https://mirrors.aliyun.com/nvidia-cuda/ubuntu1804/x86_64/7fa2af80.pub  | sudo apt-key add -
sudo chmod a+rw /etc/apt/sources.list
echo "deb https://mirrors.aliyun.com/nvidia-cuda/ubuntu1804/x86_64/ ./" > /etc/apt/sources.list
```

Before [line 25 in DockerfileGPU](https://github.com/Xilinx/Vitis-AI/blob/master/setup/docker/docker/DockerfileGPU#L25), add the following RUN statement:

```Dockerfile
RUN mv /etc/apt/sources.list.d/cuda.list /etc/apt/sources.list.d/cuda.list.BACKUP \
    && mv /etc/apt/sources.list.d/nvidia-ml.list /etc/apt/sources.list.d/nvidia-ml.list.BACKUP \
    && curl -fsSL https://mirrors.aliyun.com/nvidia-cuda/ubuntu1804/x86_64/7fa2af80.pub  | apt-key add -
    && echo "deb https://mirrors.aliyun.com/nvidia-cuda/ubuntu1804/x86_64/ ./" > /etc/apt/sources.list \
    && apt-get update -y
```