Xilinx/Vitis-AI #6: DPU TRD Vitis Flow doc reference

SpontaneousDuck, 2019-12-05:
The markdown document: https://github.com/Xilinx/Vitis-AI/blob/master/DPU-TRD/prj/Vitis/README.md

This doc references UG1360 in section 5.4 which doesn't seem to exist right now. Am I missing something?

wilderfield, 2019-12-05:
@andyluo7 , can you please comment?

anacej, 2020-01-23:
Check the UG1400, I've found all needed information there.

imrickysu, 2020-03-23:
@andyluo7 Could you update the readme?

For creating the platform, this doc could be referred: https://www.xilinx.com/html_docs/xilinx2019_2/vitis_doc/hcb1561793640896.html

The Xilinx DPU Platform release notes could be referred when creating custom platforms to support Vitis-AI: https://github.com/Xilinx/Vitis_Embedded_Platform_Source/blob/master/Xilinx_Official_Platforms/zcu102_dpu/release_notes.md

`PLATFORM_REPO_PATHS` environment variable is preferred than `SDX_PLATFORM` when adding platforms to Vitis.

qianglin-xlnx, 2020-05-11:
Hi @SpontaneousDuck,
I wonder if you have solved this issue. If not, I will submit this issue to the engineer in charge of it. If it has been resolved, please let me know and I will close this issue.
Note that if we do not receive any reply from you within 2 weeks, we will assume that this issue has been resolved and will close this issue. Thank you.

SpontaneousDuck, 2020-05-11:
I was able to fin what I needed in the documents referenced in the responses above. The documentation page I referenced has not been updated yet though and still references a non-existent document.

qianglin-xlnx, 2020-05-12:
Hi @SpontaneousDuck 
Thank you for your feedback. The issue has been fixed and README.md in https://github.com/Xilinx/Vitis-AI/blob/master/DPU-TRD/prj/Vitis/ has been updated.

closed 2020-05-28 by qianglin-xlnx