Xilinx/Vitis-AI #10: No module named 'dnnc' - vai_c_tensorflow

mKuznii, 2019-12-06:
When I try to run script **6_tf_compile.sh** from **Vitis-AI/mpsoc/vitis-ai-tool-example/** (with conda environment vitis-ai-tensorflow) the following problem occurs:
```
**************************************************
* VITIS_AI Compilation - Xilinx Inc.
**************************************************
/opt/vitis_ai/compiler/arch/dpuv2/ZCU104/ZCU104.json
Traceback (most recent call last):
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/bin/vai_c_tensorflow", line 154, in <module>
    compiler = VAI_TensorFlow_Frontend(args)
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/bin/vai_c_tensorflow", line 101, in __init__
    from dnnc.dnnc_compiler_interface import DNNCFrontend
ModuleNotFoundError: No module named 'dnnc'

```

I am using docker image xilinx/vitis-ai:tools-1.0.0-cpu.

mKuznii, 2019-12-06:
Found the solution. Copy dnnc folder to bin folder of tensorflow environment.
`cp -r /opt/vitis_ai/compiler/dnnc /opt/vitis_ai/conda/envs/vitis-ai-tensorflow/bin/`

Then new error occurs that variable VAI_ROOT does not exists. You fix it with this: 
`export VAI_ROOT=/opt/vitis_ai/`

wilderfield, 2019-12-06:
Don’t know why you are experiencing this.

Please make sure you are invoking docker_run.sh as a non-root user.

The simplest solution is:

export PYTHONPATH=/opt/vitis_ai/compiler
export VAI_ROOT=/opt/vitis_ai

closed 2019-12-06 by mKuznii