Xilinx/Vitis-AI #8: Failed to import DPU ip into vivado IP Catalog

shenjunzhong, 2019-12-06:
I try to import DPU ip into vivado IP Catalog via Settings->IP->Repository,  and I select the IP location at Vitis-AI-master/DPU-TRD/dpu_ip/dpu_eu_v3_1_0, however,  warning comes that 'Cannot get enviroment domain name variable for the component vendor name. Setting the vendor name to 'user.org'', then I failed to import the ip to the catalog. Am i do it in a wrong way? thank you.

shua1zhang, 2019-12-15:
Hi, this IP is for Vitis but not Vivado. The Vivado IP for 2019.2 will be release soon. For previous version, please see https://www.xilinx.com/products/design-tools/ai-inference/ai-developer-hub.html#edge.

shua1zhang, 2019-12-15:
> I try to import DPU ip into vivado IP Catalog via Settings->IP->Repository, and I select the IP location at Vitis-AI-master/DPU-TRD/dpu_ip/dpu_eu_v3_1_0, however, warning comes that 'Cannot get [...]

Hi, this IP is for Vitis but not Vivado. The Vivado IP for 2019.2 will be release soon. For previous version, please see https://www.xilinx.com/products/design-tools/ai-inference/ai-developer-hub.html#edge.

shua1zhang, 2019-12-22:
> I try to import DPU ip into vivado IP Catalog via Settings->IP->Repository, and I select the IP location at Vitis-AI-master/DPU-TRD/dpu_ip/dpu_eu_v3_1_0, however, warning comes that 'Cannot get [...]

Hi Shenjunzhong, 

I double check with 2019.2 released tool several time and it work fine. Are you using the latest version? Have this issue been solved? 

Thank you.

qianglin-xlnx, 2020-05-11:
Hi @shenjunzhong
I wonder if you have solved this issue. If not, I will submit this issue to the engineer in charge of it. If it has been resolved, please let me know and I will close this issue.
 Note that if we do not receive any reply from you within 2 weeks, we will assume that this issue has been resolved and will close this issue. Thank you.

qianglin-xlnx, 2020-05-28:
Hi @shenjunzhong 
Since I haven't received your reply for a long time, I will close this issue. If you still have any questions, you can reopen this issue. Thank you.

closed 2020-05-28 by qianglin-xlnx