Xilinx/Vitis-AI #9: Failed to import the vai.dpuv1 lib...

matlinsas, 2019-12-06:
Hi,
 I failed to import the vai library with the CPU container : xdock:5000/vitis-ai  1.0.0-cpu .

>>> from vai.dpuv1.tools.compile.bin.xfdnn_compiler_caffe  import CaffeFrontend as xfdnnCompiler
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ModuleNotFoundError: No module named 'vai'

Following is the $PYTHONPATH information: 

 /workspace/alveo/overlaybins/setup.sh
------------------
Using VAI_ALVEO_ROOT
------------------
/workspace/alveo

---------------------
Using LD_LIBRARY_PATH
---------------------
/opt/xilinx/xrt/lib:
-------------------
Using LIBXDNN_PATH
-------------------
/lib/libxfdnn.so

-------------------
PYTHONPATH
-------------------
/workspace/alveo:/workspace/alveo/apps/yolo:/workspace/alveo/apps/yolo/nms:/workspace/alveo/xfmlp/python:/opt/xilinx/xrt/python:/workspace/alveo:/workspace/alveo/apps/yolo:/workspace/alveo/apps/yolo/nms:/workspace/alveo/xfmlp/python:/opt/vitis_ai/compiler

---------------------
Verifying XILINX_XRT
---------------------
XILINX_XRT      : /opt/xilinx/xrt
PATH            : /opt/xilinx/xrt/bin:/opt/xilinx/xrt/bin:/opt/vitis_ai/conda/bin:/opt/vitis_ai/utility:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
LD_LIBRARY_PATH : /opt/xilinx/xrt/lib:/opt/xilinx/xrt/lib:
PYTHONPATH     : /opt/xilinx/xrt/python:/workspace/alveo:/workspace/alveo/apps/yolo:/workspace/alveo/apps/yolo/nms:/workspace/alveo/xfmlp/python:/opt/xilinx/xrt/python:/workspace/alveo:/workspace/alveo/apps/yolo:/workspace/alveo/apps/yolo/nms:/workspace/alveo/xfmlp/python:/opt/vitis_ai/compiler

Thanks,
Bean

matlinsas, 2019-12-06:
And the jupyter is not installed at this image. 

/workspace/alveo$ jupyter notebook --no-browser --ip=0.0.0.0 --NotebookApp.token='' --NotebookApp.password=''
bash: jupyter: command not found

matlinsas, 2019-12-06:
After active the caffe conda environment by following commands: 
> source activate vitis-ai-caffe

I can import above library in the python.

And I tired the caffe example at following directory: 
/workspace/alveo/examples/caffe/ssd-detect

> python run_ssd.py --prototxt /opt/models/caffe/inception_v2_ssd/inception_v2_ssd_train.prototxt --caffemodel /opt/models/caffe/inception_v2_ssd/inception_v2_ssd.caffemodel --prepare

It reports the 
FileNotFoundError: [Errno 2] No such file or directory: 'quantize_results/deploy.prototxt'

It seems that the quantize step didn't generate any results in the quantize_results.  Any suggestion? 

Thanks,
Bean.

wilderfield, 2019-12-09:
Hi @matlinsas,

In regards to your most recent post. We are looking into it. There is a problem with ssd-detect, we need to make some minor fixes to the code, and polish the instructions. We apologize for the issue. We should have it fixed by the end of the week.

@satyakee1 is working on this.

tarun28jain, 2019-12-18:
In replace_mluser.py under substitute function, after writing values to file, file handle needs to be closed so that changes are written to the file. Then again run "python replace_mluser.py --modelsdir models" after setting $VAI_ALVEO_ROOT in env and prepare command will proceed then.
After then it asks for overlays "/workspace/alveo/overlays/xdnnv3" which is not present here. Actually it is present in directory "/opt/xilinx/overlaybins/xdnnv3". Need to copy it here for prepare command to work successfully

tarun28jain, 2019-12-18:
I am facing issue in next command while running inference on entire dataset

My card is getting validated successfully
(vitis-ai-caffe) root@ubuntu:/workspace/alveo/examples/caffe/ssd-detect# /opt/xilinx/xrt/bin/xbutil validate
INFO: Found 1 cards

INFO: Validating card[0]: xilinx_u200_xdma_201830_2
INFO: == Starting AUX power connector check:
INFO: == AUX power connector check PASSED
INFO: == Starting PCIE link check:
INFO: == PCIE link check PASSED
INFO: == Starting verify kernel test:
INFO: == verify kernel test PASSED
INFO: == Starting DMA test:
Buffer Size: 256 MB
Host -> PCIe -> FPGA write bandwidth = 10438.6 MB/s
Host <- PCIe <- FPGA read bandwidth = 12147.1 MB/s
INFO: == DMA test PASSED
INFO: == Starting device memory bandwidth test:
............
Maximum throughput: 52428 MB/s
INFO: == device memory bandwidth test PASSED
INFO: == Starting PCIE peer-to-peer test:
P2P BAR is not enabled. Skipping validation
INFO: == PCIE peer-to-peer test SKIPPED
INFO: == Starting memory-to-memory DMA test:
bank0 -> bank1 M2M bandwidth: 11975.5 MB/s
bank0 -> bank2 M2M bandwidth: 12082.9 MB/s
bank0 -> bank3 M2M bandwidth: 12081.7 MB/s
bank1 -> bank2 M2M bandwidth: 12082.3 MB/s
bank1 -> bank3 M2M bandwidth: 12082.3 MB/s
bank2 -> bank3 M2M bandwidth: 12081.2 MB/s
INFO: == memory-to-memory DMA test PASSED
INFO: Card[0] validated successfully.

But on running inference with, i am getting this error
Speaking to Butler
Response from Butler is:
errCode: errCode: 18
errCode String: XCLBIN_DOWNLOAD_ERROR
myHandle: 0
valid: 1

Butler connection failed, exiting after too many errors
python: /usr/include/boost/thread/pthread/mutex.hpp:111: boost::mutex::~mutex(): Assertion `!res' failed.
Aborted (core dumped)

Any suggestions on how to proceed or what to check??

wilderfield, 2020-01-07:
This should be fixed, please open a new issue if you are still experiencing problems

closed 2020-01-07 by wilderfield