Xilinx/Vitis-AI #14: When using docker-based model fixed-point tool, what is the specific format of calib.txt file in the quantization process of caffe's target detection model?

tensorflowt, 2019-12-18:
hi,
I have configured the docker-based model quantization tool (CPU version). However, I encountered the following problems when I fixed the detection model.
[image]
When I ran a problem query in the xilinx community, I found the following solutions:
`https://forums.xilinx.com/t5/Machine-Learning/trouble-using-decent-sh/m-p/987902#M1301`
I tried it, but my problem was not solved.
I would like to know the specific form of calib.txt required for the new quantitative tool(based on caffe).
Because I tried both of the following methods and it didn't work.
>0001.jpg  1
  0002.jpg   1

or:
>0001.jpg  
  0002.jpg

shua1zhang, 2019-12-22:
Hi @tensorflowt ,

According to your output, I think the path to your images are not correctly configured. Let take files in "https://github.com/Xilinx/Vitis-AI/tree/master/mpsoc/vitis-ai-tool-example" as example. 

In this example, we assume "vai_q_caffe" is being run under path of "vitis-ai-tool-example". 
[image]

The tool will use the prototxt file in "cf_resnet50" folder and "list.txt" in "images" folder, with configuration as below:
[image]
[image]

In this correct configuration, tool will find images under folder "images/" (make sure '/' is not missing) with name specified in "list.txt". Taking first image as example, vai_q_caffe will find "./images/ILSVRC2012_val_00000001.JPEG" and its lable “65” according to the "list.txt". 

Please check if you have set the path correctly.

qianglin-xlnx, 2020-05-11:
Hi @tensorflowt 
I wonder if you have solved this issue. If not, I will submit this issue to the engineer in charge of it. If it has been resolved, please close this issue.
Note that if we do not receive any reply from you within 2 weeks, we will assume that this issue has been resolved and will close this issue. Thank you.

qianglin-xlnx, 2020-05-28:
Hi @tensorflowt 
Since I haven't received your reply for a long time, I will close this issue. If you still have any questions, you can reopen this issue. Thank you.

closed 2020-05-28 by qianglin-xlnx