Xilinx/Vitis-AI #16: caffe xilinx can not pass make runtest?

zfzfdafei, 2019-12-19:
when I make the caffe_xilinx(use cmake ..), I meet some problems, and finally I can make all successfully, but when I try to make runtest, some problem happens.Problems like "can not find files " happend when compile test_conv_bn_fixed_layer and test_crypto.So I remove the test_conv_bn_fixed_layer.cpp and test_crypto.cpp, the problems disapper.
but then, make runtest have such errors：
[----------] 1 test from LayerFactoryTest/0, where TypeParam = caffe::CPUDevice<float>
[ RUN      ] LayerFactoryTest/0.TestCreateLayer
F1219 15:40:15.665498 12839 db_leveldb.cpp:16] Check failed: status.ok() Failed to open leveldb 
IO error: /LOCK: Permission denied
*** Check failure stack trace: ***
    @     0x7f94c81335cd  google::LogMessage::Fail()
    @     0x7f94c8135433  google::LogMessage::SendToLog()
    @     0x7f94c813315b  google::LogMessage::Flush()
    @     0x7f94c8135e1e  google::LogMessageFatal::~LogMessageFatal()
    @     0x7f94c8e447b3  caffe::db::LevelDB::Open()
    @     0x7f94c9129b25  caffe::DataReader<>::Body::InternalThreadEntry()
    @     0x7f94c8de07f5  caffe::InternalThread::entry()
    @     0x7f94c879d5d5  (unknown)
    @     0x7f94c835e6ba  start_thread
    @     0x7f94bdcdf41d  clone
    @              (nil)  (unknown)
Aborted (core dumped)
src/caffe/test/CMakeFiles/runtest.dir/build.make:57: recipe for target 'src/caffe/test/CMakeFiles/runtest' failed
when I try to train the mobilenetV2 ssd in my own dataset, errors Check failed: background_label_id != label (0 vs. 0) Found background label in the dataset  happened. but use the original caffe ssd to train the same dataset, no errors. I wonder how I can train my own dataset in caffe_xilinx?

wilderfield, 2020-01-06:
Where are you getting source code for "caffe_xilinx", as far as I am aware it is not distributed. In Vitis-AI we deliver a quantizer for quantizing caffe networks, and we also give a prebuilt version of Caffe as part of the anaconda environment ( conda activate vitis-ai-caffe ).

qianglin-xlnx, 2020-05-11:
Hi @zfzfdafei 
I wonder if you have solved this issue. If not, I will submit this issue to the engineer in charge of it. If it has been resolved, please close this issue.
Note that if we do not receive any reply from you within 2 weeks, we will assume that this issue has been resolved and will close this issue. Thank you.

qianglin-xlnx, 2020-05-28:
HI @zfzfdafei 
Since I haven't received your reply for a long time, I will close this issue. If you still have any questions, you can reopen this issue. Thank you.

closed 2020-05-28 by qianglin-xlnx

deke997, 2020-06-28:
caffe-xilinx source code is here: https://www.xilinx.com/bin/public/openDownload?filename=caffe-xilinx-1.1.zip

It is listed on this page: https://github.com/Xilinx/AI-Model-Zoo#model-download

It is a requirement to run any of the models in the model zoo. When you try to run any of the tests on those models, you get:

`~/Vitis-AI/AI-Model-Zoo/models/cf_resnet50_imagenet_224_224_7.7G/code/test$ ./test.sh 
./test.sh: line 96: ../../../caffe-xilinx/build/tools/caffe: No such file or directory
`

However, this implementation of caffe-xilinx is very very broken, and it will not install as it is. I can't get it to work at all