Xilinx/Vitis-AI #17: about vai_q_caffe problem

akioolin, 2019-12-19:
Hi, 
I try to use vai_q_caffe to do a caffe model quantization. the final message is 
W1219 00:02:05.692003    91 convert_proto.cpp:1401] [DEPLOY WARNING] Layer data's output blob is all zero, this may cause error for DNNC compiler. Please check the float model.
I1219 00:02:05.696166    91 decent_q.cpp:399] Deploy Done!

Does this will has some side effect?

BR, Akio

jimheaton, 2019-12-19:
Yes,

You do need to fix this.

I just ran into the exact same error message yesterday. In my case I was not properly pointing to the calibration images.
You will typically need to edit your prototext file in order to quantize.

Here’s an example of what I had to do for tiny yolo:

[cid:image001.jpg@01D5B677.5928D7B0]

Thanks,
Jim

EXTERNAL EMAIL

akioolin, 2019-12-20:
@jimheaton  Thank you very much. for calibration images, where do I find the proper setting for vai_q_caffe to get these images for calibration?

BR, Akio

jimheaton, 2019-12-20:
The calibration images are reference in the prototext file, and the vai_q_caffe quantizer will then read them in.

What type of model do you want to quantize? Typically you would use  a subset of your validation images from training.

For our resenet50 examples we imagenet.

Regards,
Jim

EXTERNAL EMAIL

jimheaton, 2019-12-21:
@akioolin 
Here is  my example prototxt showing  @reference to calibration images.
`name: "Darkent2Caffe"
#input: "data"
#input_dim: 1
#input_dim: 3
#input_dim: 416
#input_dim: 416

####Change input data layer to COCO validation images #####
layer {
  name: "data"
  type: "ImageData"
  top: "data"
  top: "label"
  include {
    phase: TRAIN
  }
  transform_param {
    mirror: false
    yolo_height:416  #change height according to Darknet model
    yolo_width:416   #change width according to Darknet model
  }
  image_data_param {
    source: "calib.txt"  #list of calibration imaages     
    root_folder: "../darknet/data/coco/images/val2014/" #path to calibartion images

    batch_size: 1
    shuffle: false
  }
}
#####No changes to the below layers#####

layer {
    bottom: "data"
    top: "layer0-conv"
    name: "layer0-conv"
    type: "Convolution"
    convolution_param {
        num_output: 16
        kernel_size: 3
        pad: 1
        stride: 1
        bias_term: false
    }
}`

akioolin, 2019-12-24:
@jimheaton  Thank you very much!!!
the key field name is the following,
image_data_param {
source: "calib.txt" #list of calibration imaages
root_folder: "../darknet/data/coco/images/val2014/" #path to calibartion images
batch_size: 1
shuffle: false
}

I'll try to add this setting in my caffe prototxt file.

BR,Akio

qianglin-xlnx, 2020-05-11:
Hi @akioolin 
I wonder if you have solved this issue. If not, I will submit this issue to the engineer in charge of it. If it has been resolved, please close this issue.
Note that if we do not receive any reply from you within 2 weeks, we will assume that this issue has been resolved and will close this issue. Thank you.

closed 2020-05-11 by akioolin