Xilinx/Vitis-AI #18: Failed to Load&Run Docker Container

louislxw, 2019-12-23:
Hi, 

I am trying to install and run the vitis-ai-docker-tools following the instructions under "Getting Started". I was able to install docker and put my linux user in the group docker. However, there is an issue when I ran the command "./docker_run.sh xilinx/vitis-ai:tools-1.0.0-cpu". Error messages:

docker: unknown server OS: .
See 'docker run --help'.

Could anyone support with this issue? Thanks! 

Regards, 
Louis

closed 2020-01-06 by louislxw

AlbertaBeef, 2020-01-20:
Louis,
I am running into the same issue, did you find a fix ?  can you share ?
Mario.

louislxw, 2020-01-20:
> Louis,
> I am running into the same issue, did you find a fix ? can you share ?
> Mario.

Hi Mario, 

I followed the instructions carefully twice and solved the previous issue. 
I guess you were able to install docker with no issue. Be careful with the 2nd step "Ensure your linux user is in the group docker" (https://docs.docker.com/install/linux/linux-postinstall/). Specifically, verify that you can run the following command **without sudo**
 $ docker run hello-world

Regards,
Louis

AlbertaBeef, 2020-01-20:
Thank you Louis, that fixed the issue.

rmsys, 2020-11-12:
I know this post is old but maybe help somebody. 

**if you try:**

```
ubuntu@ip:/$ docker run hello-world
docker: Got permission denied while trying to connect to the Docker daemon socket at unix:///var/run/docker.sock: Post http://%2Fvar%2Frun%2Fdocker.sock/v1.40/cont.
```

**so, update permission:**
`ubuntu@ip:/$ sudo chmod 666 /var/run/docker.sock
`

hanxue, 2020-11-12:
@rmsys good workaround. Changing the permissions of /var/run/docker.sock by hand will only fix the problem temporarily. The permissions of `/var/run/docker.sock` will be reset when the Docker daemon is next reset. 

A more permanent solution is to create the `docker` group on your host machine and add the users running docker to the `docker` group per https://docs.docker.com/install/linux/linux-postinstall/

mo979, 2021-05-05:
I met the problem and added sudo before the shell command. Then it ran successfully.

tan24012, 2023-02-10:
aa
[Makefile.zip](https://github.com/Xilinx/Vitis-AI/files/10704847/Makefile.zip)

gregpalaci, 2023-09-30:
https://github.com/docker/for-mac/issues/6531#issuecomment-1442392362