Xilinx/Vitis-AI #21: build adas_detection error

runner0621, 2019-12-24:
build adas_detection meet cannot find -lhineon/ln2cube error,  compile the model on the board directly

module path: Vitis-AI/mpsoc/vitis_ai_samples_zcu104/adas_detection

error info:
/usr/lib/gcc/aarch64-xilinx-linux/8.2.0/../../../../aarch64-xilinx-linux/bin/ld: cannot find -lhineon
/usr/lib/gcc/aarch64-xilinx-linux/8.2.0/../../../../aarch64-xilinx-linux/bin/ld: cannot find -ln2cube

qianglin-xlnx, 2020-05-11:
Hi @runner0621 
I wonder if you have solved this issue. If not, I will submit this issue to the engineer in charge of it. If it has been resolved, please close this issue.
Note that if we do not receive any reply from you within 2 weeks, we will assume that this issue has been resolved and will close this issue. Thank you.

qianglin-xlnx, 2020-05-28:
@runner0621 
I think you miss the following step. Could you please check it again? Let me know if you still have any problems. Thank you. 
[image]

qianglin-xlnx, 2020-06-15:
Hi @runner0621 
Since I haven't received your reply for a long time, I will close this issue. If you still have a problem, you can reopen it. Thank you very much.

closed 2020-06-15 by qianglin-xlnx