Xilinx/Vitis-AI #23: #!/bin/sh TF_NETWORK_PATH=tf_yolov3_voc_416_416_65.63G vai_q_tensorflow quantize --input_frozen_graph tf_yolov3_voc_416_416_65.63G/float/yolov3_voc.pb --input_fn input_fn.calib_input --output_dir vai_q_output --input_nodes input_1 --output_nodes conv2d_59/BiasAdd,conv2d_67/BiasAdd,conv2d_75/BiasAdd  --input_shapes ?,416,416,3  --calib_iter 50

zyxcambridge, 2019-12-27:
#!/bin/sh
TF_NETWORK_PATH=tf_yolov3_voc_416_416_65.63G
vai_q_tensorflow quantize --input_frozen_graph tf_yolov3_voc_416_416_65.63G/float/yolov3_voc.pb --input_fn input_fn.calib_input --output_dir vai_q_output --input_nodes input_1 --output_nodes conv2d_59/BiasAdd,conv2d_67/BiasAdd,conv2d_75/BiasAdd  --input_shapes ?,416,416,3  --calib_iter 50

alib_data/COCO_train2014_000000000036.jpg
 10% (5 of 50) |#####################                                                                                                                                                                                                   | Elapsed Time: 0:00:27 ETA:   0:04:07calib_data/COCO_train2014_000000000049.jpg
Traceback (most recent call last):
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/tensorflow/python/client/session.py", line 1334, in _do_call
    return fn(*args)
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/tensorflow/python/client/session.py", line 1319, in _run_fn
    options, feed_dict, fetch_list, target_list, run_metadata)
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/tensorflow/python/client/session.py", line 1407, in _call_tf_sessionrun
    run_metadata)
tensorflow.python.framework.errors_impl.InvalidArgumentError: output dimensions must be positive
	 [[{{node up_sampling2d_1/ResizeNearestNeighbor}} = ResizeNearestNeighbor[T=DT_FLOAT, align_corners=false, _device="/job:localhost/replica:0/task:0/device:CPU:0"](leaky_re_lu_59/sub/aquant, up_sampling2d_1/mul)]]

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/bin/vai_q_tensorflow", line 10, in <module>
    sys.exit(run_main())
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/tensorflow/contrib/decent_q/python/decent_q.py", line 744, in run_main
    app.run(main=my_main, argv=[sys.argv[0]] + unparsed)
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/tensorflow/python/platform/app.py", line 125, in run
    _sys.exit(main(argv))
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/tensorflow/contrib/decent_q/python/decent_q.py", line 743, in <lambda>
    my_main = lambda unused_args: main(unused_args, FLAGS)
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/tensorflow/contrib/decent_q/python/decent_q.py", line 539, in main
    flags.skip_check)
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/tensorflow/contrib/decent_q/python/decent_q.py", line 327, in quantize_frozen
    q_config, s_config)
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/tensorflow/contrib/decent_q/python/decent_q.py", line 276, in calibrate_frozen
    sess.run(output_tensors, feed_dict)
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/tensorflow/python/client/session.py", line 929, in run
    run_metadata_ptr)
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/tensorflow/python/client/session.py", line 1152, in _run
    feed_dict_tensor, options, run_metadata)
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/tensorflow/python/client/session.py", line 1328, in _do_run
    run_metadata)
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/tensorflow/python/client/session.py", line 1348, in _do_call
    raise type(e)(node_def, op, message)
tensorflow.python.framework.errors_impl.InvalidArgumentError: output dimensions must be positive
	 [[node up_sampling2d_1/ResizeNearestNeighbor (defined at /opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/tensorflow/contrib/decent_q/python/decent_q.py:260)  = ResizeNearestNeighbor[T=DT_FLOAT, align_corners=false, _device="/job:localhost/replica:0/task:0/device:CPU:0"](leaky_re_lu_59/sub/aquant, up_sampling2d_1/mul)]]

Caused by op 'up_sampling2d_1/ResizeNearestNeighbor', defined at:
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/bin/vai_q_tensorflow", line 10, in <module>
    sys.exit(run_main())
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/tensorflow/contrib/decent_q/python/decent_q.py", line 744, in run_main
    app.run(main=my_main, argv=[sys.argv[0]] + unparsed)
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/tensorflow/python/platform/app.py", line 125, in run
    _sys.exit(main(argv))
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/tensorflow/contrib/decent_q/python/decent_q.py", line 743, in <lambda>
    my_main = lambda unused_args: main(unused_args, FLAGS)
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/tensorflow/contrib/decent_q/python/decent_q.py", line 539, in main
    flags.skip_check)
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/tensorflow/contrib/decent_q/python/decent_q.py", line 327, in quantize_frozen
    q_config, s_config)
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/tensorflow/contrib/decent_q/python/decent_q.py", line 260, in calibrate_frozen
    tf.import_graph_def(calib_graph_def, name='')
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/tensorflow/python/util/deprecation.py", line 488, in new_func
    return func(*args, **kwargs)
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/tensorflow/python/framework/importer.py", line 442, in import_graph_def
    _ProcessNewOps(graph)
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/tensorflow/python/framework/importer.py", line 234, in _ProcessNewOps
    for new_op in graph._add_new_tf_operations(compute_devices=False):  # pylint: disable=protected-access
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/tensorflow/python/framework/ops.py", line 3440, in _add_new_tf_operations
    for c_op in c_api_util.new_tf_operations(self)
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/tensorflow/python/framework/ops.py", line 3440, in <listcomp>
    for c_op in c_api_util.new_tf_operations(self)
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/tensorflow/python/framework/ops.py", line 3299, in _create_op_from_tf_operation
    ret = Operation(c_op, self)
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/tensorflow/python/framework/ops.py", line 1770, in __init__
    self._traceback = tf_stack.extract_stack()

InvalidArgumentError (see above for traceback): output dimensions must be positive
	 [[node up_sampling2d_1/ResizeNearestNeighbor (defined at /opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/tensorflow/contrib/decent_q/python/decent_q.py:260)  = ResizeNearestNeighbor[T=DT_FLOAT, align_corners=false, _device="/job:localhost/replica:0/task:0/device:CPU:0"](leaky_re_lu_59/sub/aquant, up_sampling2d_1/mul)]]

A1exy, 2020-01-08:
Looks like the document yolov3 supports caffe, does tensorflow yolov3 not support it?

[image]

qianglin-xlnx, 2020-05-28:
@zyxcambridge 
As it has been a long time, I don't know if your issue has been solved. If not, please let me know and describe in detail the steps to reproduce the issue. If possible, I would suggest you to try it again with Vai1.1. Thank you.

qianglin-xlnx, 2020-06-15:
Hi @zyxcambridge 
Since I haven't received your reply for a long time, I will close this issue. If you still have a problem, you can reopen it. Thank you very much.

closed 2020-06-15 by qianglin-xlnx

Lavanya-deepedge, 2020-12-18:
Hi @qianglin-xlnx

I am facing the same problem could you please help with those Errors