Xilinx/Vitis-AI #24: tensorflow.python.framework.errors_impl.NotFoundError: Op type not registered 'LeakyRelu' in binary running on cb56f6cc53b2. Make sure the Op and Kernel are registered in the binary running in this process. Note that if you are loading a saved graph which used ops from tf.contrib, accessing (e.g.) `tf.contrib.resampler` should be done before importing the graph, as contrib ops are lazily registered when the module is first accessed.

zyxcambridge, 2019-12-30:
tensorflow.python.framework.errors_impl.NotFoundError: Op type not registered 'LeakyRelu' in binary running on cb56f6cc53b2. Make sure the Op and Kernel are registered in the binary running in this process. Note that if you are loading a saved graph which used ops from tf.contrib, accessing (e.g.) `tf.contrib.resampler` should be done before importing the graph, as contrib ops are lazily registered when the module is first accessed.

foolmarks, 2019-12-30:
Make sure to import the contrib.decent_q module:

import tensorflow.contrib.decent_q

zyxcambridge, 2019-12-30:
I  have import ，the problem is  about the tensorflow version ，xilinx  version is 1.12，my model for yolov3 is  1.13.1

closed 2019-12-30 by zyxcambridge