Xilinx/Vitis-AI #27: Questions about yolov3 training code

A1exy, 2020-01-08:
Hi, I want to train yolov3 model and deploy it on Xilinx dnndk，but i don't have the caffe training source code of yolov3,  can you provide source code or some references?

jimheaton, 2020-01-09:
You can train with Darknet, use the Xlinix Darknet 2 Caffe conversion flow, and then use the Caffe version of dnnc.

The procedure is described in detail on our Darknet 2 Caffe Tutorial, however this has been temporally removed from the Xilinx GitHub site pending some internal reviews.

Please check back soon at https://github.com/Xilinx/Edge-AI-Platform-Tutorials

EXTERNAL EMAIL

A1exy, 2020-01-13:
@jimheaton Thanks for your reply, and when do you expect to release Darknet 2 Caffe Tutorial.
Best Regards!

qianglin-xlnx, 2020-05-11:
Hi @A1exy 
I wonder if you have solved this issue. If not, I will submit this issue to the engineer in charge of it. If it has been resolved, please close this issue.
Note that if we do not receive any reply from you within 2 weeks, we will assume that this issue has been resolved and will close this issue. Thank you.

A1exy, 2020-05-11:
@qianglin-xlnx 
Hi, I haven't solved the problem yet,  because don't have the script for the conversion of your Darknet 2 Caffe Tutorial.

qianglin-xlnx, 2020-05-11:
@A1exy 
Hi, please visit the following page and download the caffe-xilinx.zip packet. Unzip it and you will find the convert.py script under /scripts.
 https://github.com/Xilinx/AI-Model-Zoo/tree/1387830ef9b846255245c1dc063e7c86ac71498e 

Or you can directly download the caffe-xilinx.zip from the following link.
https://www.xilinx.com/bin/public/openDownload?filename=caffe-xilinx-1.1.zip

A1exy, 2020-05-11:
@qianglin-xlnx 
OK，thanks,  I'll try it later.

closed 2020-05-11 by A1exy

HeXinya, 2020-06-03:
Hi,

The darknet2caffe tutorial mentioned anove is no long available... Any idea of alternative repository? Thanks!

jimheaton, 2020-06-03:
You can use the following flow:

1.  Use  https://github.com/qqwweee/keras-yolo to<https://github.com/qqwweee/keras-yolo%C2%A0to> transform from darknet to keras
2 . Use https://github.com/Xilinx/Vitis-AI-Tutorials/tree/Keras-Freeze-with-Vitis-AI method<https://github.com/Xilinx/Vitis-AI-Tutorials/tree/Keras-Freeze-with-Vitis-AI%C2%A0method> 1 to convert to TensorFlow and freeze model.
3.  Then you can take the TensorFlow frozen model in Vitis-AI.

 https://github.com/qqwweee/keras-yolo3