Xilinx/Vitis-AI #29: Confusion about the model：cf_fpn_cityscapes_256_512_8.9G

tensorflowt, 2020-01-10:
hi，I've been training cf_fpn_cityscapes_256_512_8.9G models myself recently.
Because of the problem of my hardware resources（2080ti）, when I trained the dataset（cityscapes）,batch size could only be set to 1.
However, when batch size=1, the loss value of the model remains at 2.9 and does not decrease.
So I wanted to try to set a larger batch size by reducing the image size（from 1024x2048 to 256x512） of the dataset.As follows：
[image]
But when I used the scaled atlas to train FPN, I encountered the following bug：
[image]
Therefore, I am confused. The questions I am not sure about are as follows：
1.If the input size of the model is 256*512, it is ok for me to scale the image of the training, but why is the error reported？
2.What is the meaning of line 77of the file：“./cf_fpn_cityscapes_256_512_8.9G/code/ReadMe.md”?
[image]

qianglin-xlnx, 2020-05-12:
Hi @tensorflowt 
I wonder if you have solved this issue. If not, please describe your steps in detail and I will submit this issue to the engineer in charge of it. If it has been resolved, please close this issue.
Note that if we do not receive any reply from you within 2 weeks, we will assume that this issue has been resolved and will close this issue. Thank you.

wanglixilinx, 2020-05-28:
Hi @tensorflowt,
how do you scaled the data?  please check you processed annotation mask to make sure the pixel value is among [0, 18] except the ignore id 255, you need use the nearest mode to resize the annotation,

qianglin-xlnx, 2020-06-15:
Hi @tensorflowt 
Since I haven't received your reply for a long time, I will close this issue. If you still have a problem, you can reopen it. Thank you very much.

closed 2020-06-15 by qianglin-xlnx