Xilinx/Vitis-AI #30: link in Vitis-AI/Vitis-AI-Library/README is outdated.

HsimWong, 2020-01-13:
A link is given in the quick start session in README.md which requires users to download the vitis-ai-docker-runtime image. The link is not accessible, returning a 404 page.

Mpic, 2020-01-27:
What specific link are you talking about ? Cloning the git repository as instructed in README.md and launching `Vitis-AI/docker/docker_run.sh xilinx/vitis-ai:runtime-1.0.0-cpu` as told in [load_run_docker.md](https://github.com/Xilinx/Vitis-AI/blob/master/doc/install_docker/load_run_docker.md) automatically downloads and launches the vitis-ai-docker-runtime image for me.

HsimWong, 2020-01-29:
I think I have clearly stated in my **humble** description. 
The link is given in **Vitis-AI/Vitis-AI-Library/README**,
and there is a section called **quick start** in this file,
which gives a link naming vitis-ai-docker-runtime.
It is returning 404 now.

But it's okay, I'll raise a code comment.

Mpic, 2020-01-29:
Oh, I see. The docker image is there : https://hub.docker.com/r/xilinx/vitis-ai/tags and I'm not sure if you really have to download it manually as it is on Docker Hub, it's easier to just give it as argument to `docker_run.sh`.

`/Vitis-AI-Library/README.md` seems outdated indeed.

HsimWong, 2020-01-29:
Thank you very much. I just happen to find that it's no longer valid.

closed 2020-01-29 by HsimWong