Xilinx/Vitis-AI #31: Black screen when running examples

Morett07, 2020-01-14:
Hello,

I have a zcu 104 and I try to start examples as it is explain here : 

https://github.com/Xilinx/Vitis-AI/blob/master/mpsoc/runtime_docker.md

I flashed the sd card with this img : xilinx-zcu104-dpu-v2019.2.img.gz and Etcher

then I successfully installed xilinx_vai_board_package like it is described in Vitis user guide p26 : 

https://www.xilinx.com/support/documentation/sw_manuals/vitis_ai/1_0/ug1414-vitis-ai.pdf#unique_34

I managed to cross-compile applications in runtime docker, I tried the following examples : 

    - from dnndk : 
        - resnet50
        - adas_detection
        - segmentation

    - from dnndk : 
        - adas_detection
        - segmentation

For all of these examples I obtain a black screen and the the board freezes. 

I tried standalone connection (Display port) and ssh.

I haven't any signals on HDMI tx.

By ssh a black pop up appears and the board  freezes.

[image]

I tried to see kernel log but I see nothing suspicious.

[image]

Did miss something on the setup ?

Best regards,

Matthieu

mullerrwd, 2020-02-24:
I can second the problems described by @Morett07. See issue #45.

qianglin-xlnx, 2020-05-11:
Hi @Morett07 
I wonder if you have solved this issue. If not, I will submit this issue to the engineer in charge of it. If it has been resolved, please close this issue.
Note that if we do not receive any reply from you within 2 weeks, we will assume that this issue has been resolved and will close this issue. Thank you.

qianglin-xlnx, 2020-05-11:
Note: You may need to run command irps5401 to trigger the power management patch for ZCU104 to avoid system hang or power off issue when running Vitis™ AI examples.

qianglin-xlnx, 2020-05-28:
Hi @Morett07  The following issue is same as you and it's solved. You can find the solution here. Let me know if you still have any problem.
https://github.com/Xilinx/Vitis-AI/issues/45

qianglin-xlnx, 2020-06-15:
Hi @Morett07 
Since I haven't received your feedback for a long time, I will close this issue.If you still have a question, you can reopen it. Thank you

closed 2020-06-15 by qianglin-xlnx