Xilinx/Vitis-AI #32: Profiler only showing Conv2D Layers

asharma20, 2020-01-14:
I am trying to retrieve the N2Cube Profile results for each layer of the MobilenetV2 tensorflow model that I downloaded from Vitis-AI/AI-Model-Zoo. I expect it to show DepthwiseConv layers and an AvgPool layer, but it seems to only return Conv2D layers:
[image]

The output from `dexplorer -w`:
[image]

Here are the steps to recreate the issue:
1. Compile all_models/tf_mobilenetv2_imagenet_224_224_1.17G/fix/deploy_model.pb
`dnnc --parser=tensorflow --frozen_pb=fix/deploy_model.pb --output_dir=xilinx_modelzoo --dcf=../../../dcf/ZCU104.dcf --mode=debug --cpu_arch=arm64 --net_name=MobilenetV2 --save_kernel --dump=all`
2. Copy over the generated ELF file to the ZCU104
3. Create a Makefile and main.cc similar to the files in ZCU104/samples/resnet50, updated with the correct DPU Kernel Name, Input Node, and Output Node
4. Run with profile mode
`dexplorer -m profile`
5. Compile main.cc linked with ELF and run the generated executable

Is there a way to get the profiler to generate these results with each layer in the network?

qianglin-xlnx, 2020-05-13:
Hi @asharma20 
Our compiler will do some layer fusion optimization to improve performance and will try to use on-chip memory to save feature map as long as we can. For example, if possible, the conv layer following by a pooling layer(convolution+pool) will be fused into a special combined layer, which we call it a 'super layer'. The output feature-map of conv will be all saved in on-chip bank and only the output of pool will be saved in DDR. It's the same case if convolution+depthwise can be fused too.
For the dpu runtime, it will treate each 'super' layer as a single scheduling unit and schedule the 'super layer' one by one(in debug mode) in dpu and there is no need for the runtime to know the details about the fusion info.
In summay, the profiler is showing the results for 'super layer'. If some layers are fused with other, they will not be visiable to runtime and profiler.
Hope it helps, thanks very much.

qianglin-xlnx, 2020-05-28:
hi @asharma20 
Since I haven't received your reply for a long time, I will close this issue. If you still have any questions, you can reopen this issue. Thank you.

closed 2020-05-28 by qianglin-xlnx