Xilinx/Vitis-AI #33: xbutler issue while running alveo samples

mahinlma, 2020-01-17:
Installed provided xbutler package.(under alveo/package/ubuntu) 
How to solve this error?

```
Speaking to Butler 
Response from Butler is: 
errCode: errCode: 3
errCode String: NO_FREE_RESOURCE
myHandle: 0
valid: 1
```

sumitn-xilinx, 2020-01-23:
You can try restarting butler. Following are the commands (to be run outside docker).
Start: % sudo service xbutler start
Stop: % sudo service xbutler stop
Restart: % sudo service xbutler Restart

Even after that if the problem persists, you can try running following (inside docker) to get more info:
xbutler_backdoor.sh -b

mahinlma, 2020-01-28:
@sumitn-xilinx

closed 2020-01-28 by mahinlma

Hikaru-Furuta, 2021-01-07:
After restarting, I suffer this error.
Could you help me solve this one?

~~~~~~~~~~~~~~
Speaking to Butler 
Response from Butler is: 
errCode: errCode: 9
errCode String: PROGRAM_CREATE_ERROR
myHandle: 0
valid: 1
~~~~~~~~~~~~~~

hanxue, 2021-01-07:
Hi @Hikaru-Furuta , could you please raise a new issue, and together with the exact version of Vitis AI docker container that you are using?

sumitn-xilinx, 2021-01-07:
Hi @Hikaru-Furuta  - Did you try the workaround mentioned above?

Hikaru-Furuta, 2021-01-07:
@sumitn-xilinx 
Yes, I tried.
as for `xbutler_backdoor.sh -b`, got followings
~~~~~~~~~~~~~
xbutler_backdoor.sh -b
-------------------------------
Validating Arguments... passed!
-------------------------------

----------------------
Verifying CONDA_PREFIX
----------------------
----------------------
Verifying XILINX_XRT
----------------------
Pass!
Password Correct!
-----------------------
 Handle/Resource Pairs
-----------------------

-----------------------
  Service/Handle Pairs
-----------------------

-----------------------
       Port Info
-----------------------
~~~~~~~~~~~~~

sumitn-xilinx, 2021-01-07:
Thanks for sharing the output of backdoor. I deleted my comment because I saw that you faced this issue after restart only. 
This error is mostly related to XCLBIN not present or not being readable. 
Which alveo example are you trying? Can you check if the corresponding xclbin is readable?

Hikaru-Furuta, 2021-01-07:
@sumitn-xilinx 
I'm using alveo U250.
And codes I had tried is follows.

***FPGA writing command***
~~~~~~~~~~~~~
cd ~/git/Vitis13/work/alveo_xclbin-1.3.0/U250-V3INT8
/opt/xilinx/xrt/bin/xbutil program -d 0 -p dpdpuv3_wrapper.hw.xilinx_u250_xdma_201830_2.xclbin
~~~~~~~~~~~~~

***environment setup***0
~~~~~~~~~~~~~
source /opt/xilinx/xrt/setup.sh
sudo /opt/xilinx/xrt/bin/xbmgmt flash --scan
xbutil reset
~~~~~~~~~~~~~

***docker run(vai1.3)***
~~~~~~~~~~~~~
cd ~/git/Vitis13/
./docker_run.sh xilinx/vitis-ai-cpu:latest
cd /workspace/examples/DPUCADX8G/vitis_ai_alveo_samples/video_analysis
conda activate vitis-ai-tensorflow
make clean
make
export VAI_ALVEO_ROOT=/workspace/examples
./run.sh XXX.mp4
~~~~~~~~~~~~~

VishalX, 2021-01-07:
@Hikaru-Furuta 
I see a few issues with your approch.
- I see you are programming XCLBIN manually onto the card, which is DPUv3int8 (DPUCADF8H) XCLBIN, then in the env setup step, you reset the card with `xbutil reset`, which will reset the card.
- Another issue is, As I mentioned, you are programming DPUCADF8H xclbin onto the card but when you are trying the example, you're inside DPUCADX8G examples `cd /workspace/examples/DPUCADX8G/vitis_ai_alveo_samples/video_analysis`.

Can you please let us know which DPU are you targetting for your runs? 

DPUCADX8G (dpuv1) is the older DPU with Batch-1
DPUCADF8H (dpuv3int8) in the newer DPU with Batch-4

Hikaru-Furuta, 2021-01-07:
@VishalX
Thank you for your kind attention.
I'm targetting DPUCADX8G
So you mean DPUCADX8G is no longer supported by XCLBIN in vai1.3?

VishalX, 2021-01-07:
No, both are supported with VAI-1.3. 

Here are some pointers.

**DPUCADX8G Setup:**
https://github.com/Xilinx/Vitis-AI/blob/master/setup/alveo/u200_u250/README.md#setup-for-alveo

This script: https://github.com/Xilinx/Vitis-AI/blob/master/setup/alveo/u200_u250/packages/install.sh,  will do all you need for setting up for DPUCADX8G, including XCLBIN installation. You can go through the script and use the links to download packages manually as well.

**DPUCADX8G Examples:**
https://github.com/Xilinx/Vitis-AI/tree/master/examples/DPUCADX8G

Hikaru-Furuta, 2021-01-07:
@VishalX
I could compile succesfully and owe this to you
Thank you so much.