Xilinx/Vitis-AI #37: Installation failed

shijinbose, 2020-01-29:
Hi Team,
We are facing an issue while installing the Vitis-AI in our system. When we run ./docker_build.sh from Load&Run Docker Container section, The system through an error message every time. Please find the screenshot of the error message. 
[image]
 We have tried on both ubuntu18.04 and 16.04. 
Please guide me to resolve the issue.
Thanks & Regards,
Shijin Bose
shijinbose@gmail.com

qianglin-xlnx, 2020-05-11:
Hi @shijinbose 
I wonder if you have solved this issue. If not, I will submit this issue to the engineer in charge of it. If it has been resolved, please close this issue.
Note that if we do not receive any reply from you within 2 weeks, we will assume that this issue has been resolved and will close this issue. Thank you.

qianglin-xlnx, 2020-10-09:
Hi @shijinbose 
I'm going to close this. Please feel free to reopen if there are additional issues. Thank you.

closed 2020-10-09 by qianglin-xlnx