Xilinx/Vitis-AI #38: Why is there no example for zcu106？

DivineZeng, 2020-01-30:


andyluo7, 2020-03-31:
ZCU106 is not supported. Please use ZCU104 instead.

closed 2020-03-31 by andyluo7

kevinkit, 2020-08-10:
Is there any plan from XILINX to support the ZCU106?

jimheaton, 2020-08-10:
> CAUTION: This message has originated from an External Source. Please use proper judgment and caution when opening attachments, clicking links, or responding to this email.

There is no official plan to support Vitis-AI on the ZCU106.
However I have created  a port of the ZCU104 DPU TRD to ZCU106: https://github.com/jimheaton/Vitis-AI-DPU_TRD-for-ZCU106

This uses the Vitis Flow for the DPU, and requires Vitis 2019.2 and supports the Vitis-AI 1.1 release.

kevinkit, 2020-08-12:
Thank you! 

Just out of curiosity: is there a specific reason why the ZCU106 will not be officially supported and is not planned to be supported?

Greetings