Xilinx/Vitis-AI #40: Where to get the arch.json for zynq 7010 when use Vitis AI Compiler ?

zuokanmingyuli, 2020-02-03:
hi,
I have configured the docker-based model quantization tool (CPU version). However, I encountered the following problems when I use the Vitis AI Compiler tools.
The commond 'vai_c_tensorflow' need a parameter --arch arch.json. But for zynq 7010，where can I get the arch.json ?  Or is there any document for edit it ?

qianglin-xlnx, 2020-05-13:
Hi @zuokanmingyuli 
I am afraid you have to create the zynq7010.json file by yourself. You can refer to ZCU102.json to create your zynq7010.json. If you use Vai1.1, you will find it under /opt/vitis_ai/compiler/arch/dpuv2/ZCU102 when you run the docker container. 

[image]

qianglin-xlnx, 2020-12-11:
HI @zuokanmingyuli 
Since I haven't received your reply for a long time, I'm going to close this issue. If you still have any questions, please fell free to reopen it. Thank you very much.

closed 2020-12-11 by qianglin-xlnx