Xilinx/Vitis-AI #41: How to generate pp_pipeline.so for running x+ml app

mahinlma, 2020-02-04:
Hi, @wilderfield @sumitn-xilinx 
Generated xclbin for x+ml (preprossing resize operation) with L3 vision.
For running x+ml application, please provide more details regarding, how to integrate preprocessing and ml acceleraion. 
pp_pipeline.so not found.

sumitn-xilinx, 2020-02-04:
You can try following:
**Building and Running the Application**
cd $VAI_ALVEO_ROOT/apps/xplusml/src
Run make to generate the shared library file
**Running the Application**
cd $VAI_ALVEO_ROOT/apps/xplusml
Use run.sh file to run the application. Familiarize yourself with the script usage by ./run.sh -h
**Examples:**
To run image classification using Googlenet_v1
$ ./run.sh
To run image classification using resnet-50
$ ./run.sh -m resnet50

mahinlma, 2020-02-04:
@sumitn-xilinx ,
Thank you, I followed your instruction.
while running run.sh, getting this info *Device Not Supported* and not getting result/response from device. 

```
Running:
 Test: test_classify
 Model: resnet50
 Xclbin: /opt/xilinx/overlaybins/xdnnv3
 Accelerator: 0

python test_classify.py --xclbin /opt/xilinx/overlaybins/xdnnv3 --netcfg ./data/resnet50_8b_autoAllOpt.json --weights ./data/resnet50_data.h5 --labels ./synset_words.txt --quantizecfg ./data/resnet50_8b_xdnnv3.json --img_input_scale 1.0 --batch_sz 1 --vitis_rundir /tmp/vitis-ifpga/306 --images dog.jpg --deviceid 0
Found Platform
Platform Name: Xilinx
INFO: Reading /opt/xilinx/overlaybins/xdnnv3/xdnn_v3_96x16_2pe_8b_9mb_bank03.xclbin
Loading: '/opt/xilinx/overlaybins/xdnnv3/xdnn_v3_96x16_2pe_8b_9mb_bank03.xclbin'
-------------------
Speaking to Butler 
**Device Not Supported**
[XPLUSML]  Unable to Create handle for the pre-processing kernel
Response from Butler is: 
errCode: errCode: 0
errCode String: SUCCESS
myHandle: 3
valid: 1
[XDNN] loading xclbin settings from /opt/xilinx/overlaybins/xdnnv3/xdnn_v3_96x16_2pe_8b_9mb_bank03_2.xclbin.json
[XDNN] using custom DDR banks 0,3
Path /tmp/vitis-ifpga/306/weights.h5 is a file.
Loading weights/bias/quant_params to FPGA...

[XRT]    git hash                   : 192e706aea53163a04c574f9b3fe9ed76b6ca471
[XDNN]   git hash                   : 1247bc4c889955a653bfefd1777c3aa2db3b4a4e
[XDNN] kernel configuration
[XDNN]   num cores                  : 2
[XDNN]   dsp array width            : 96
[XDNN]   axi data width (in 32bits) : 16
[XDNN]   img mem size               : 9 MB
[XDNN]   max instr num              : 1536
[XDNN]   max xbar entries           : 4096
[XDNN]   version                    : 3.2
[XDNN]   8-bit mode                 : 1

```

mahinlma, 2020-02-04:
Fixed the issue. Thank you @sumitn-xilinx

mahinlma, 2020-02-04:
`[XRT] ERROR: kernel 'pp_pipeline_accel' not found.`
@sumitn-xilinx , Please tell me which xclbin, I have to load. my xdma version - xilinx_u200_xdma_201830_2

sumitn-xilinx, 2020-02-05:
You need not build the xclbin. It's part of the Vitis AI docker. If you use the run.sh from $VAI_ALVEO_ROOT/apps/xplusml, it would pick the xclbin from /opt/xilinx/overlaybins/xdnnv3/xplusml/classification/

And you should be is using "xilinx_u200_xdma_201830_1 DSA".

Let me know if you see any build errors while doing the step that I mentioned earlier ("Building and Running the Application").

mahinlma, 2020-02-05:
Hi @sumitn-xilinx , Thank you.

I'm currently using - DSA xilinx_u200_xdma_201830_2
XRT 2019.2

so please let me know, XRT version also

sumitn-xilinx, 2020-02-05:
XRT version is correct - 2019.2

And as I mentioned in my previous comment, if you face issue while building, you may see "missing cl2.hpp" error, which can be fixed by using this file from:
https://github.com/KhronosGroup/OpenCL-CLHPP/blob/master/include/CL/cl2.hpp

mahinlma, 2020-02-05:
@sumitn-xilinx Thank you, issue related cl2.hpp is fixed already. I'm facing issue while runing ./run.sh.

because of my DSA xilinx_u200_xdma_201830_2, getting Device Not Supported.

sumitn-xilinx, 2020-02-05:
Did you use the setup instructions from here:
https://github.com/Xilinx/Vitis-AI/tree/master/alveo#setup-for-alveo

Because the install.sh script mentioned will install the right version of DSA.

mahinlma, 2020-02-05:
installed same version (xilinx-u200-xdma-201830.2-2580015_16.04.deb) mentioned in the install.sh
DSA - xilinx_u200_xdma_201830_2

sumitn-xilinx, 2020-02-06:
Sorry, I should have been clear. 
X+ML example's xclbin needs "xilinx_u200_xdma_201830_1" DSA.

By default, install.sh give you "xilinx_u200_xdma_201830_2" DSA which is not good for X+ML example. It can be downloaded from here:
https://www.xilinx.com/html_docs/xilinx2019_2/vitis_doc/lhp1570811049711.html?hl=xilinx_u200_xdma_201830_1

mahinlma, 2020-02-06:
@sumitn-xilinx , Thank you. 
Installed DSA - xilinx_u200_xdma_201830_1 
getting following error
 [XRT] ERROR: kernel 'pp_pipeline_accel' not found

```

---------------------

Verifying XILINX_XRT

---------------------

XILINX_XRT      : /opt/xilinx/xrt

PATH            : /opt/xilinx/xrt/bin:/opt/xilinx/xrt/bin:/opt/vitis_ai/conda/envs/vitis-ai-caffe/bin:/opt/vitis_ai/conda/condabin:/opt/vitis_ai/conda/bin:/opt/vitis_ai/utility:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin

LD_LIBRARY_PATH : /opt/xilinx/xrt/lib:/opt/xilinx/xrt/lib:

PYTHONPATH     : /opt/xilinx/xrt/python:/opt/xilinx/xrt/python:/opt/vitis_ai/compiler

--- XCLBIN Found---

Running:

Test: test_classify

Model: googlenet_v1

Xclbin: /opt/xilinx/overlaybins/xdnnv3

Accelerator: 0

python test_classify.py --xclbin /opt/xilinx/overlaybins/xdnnv3 --netcfg ./data/googlenet_v1_8b_autoAllOpt.json --weights ./data/googlenet_v1_data.h5 --labels ./synset_words.txt --quantizecfg ./data/googlenet_v1_8b_xdnnv3.json --img_input_scale 1.0 --batch_sz 1 --vitis_rundir /tmp/vitis-ifpga/89 --images dog.jpg --deviceid 0

Found Platform

Platform Name: Xilinx

INFO: Reading /opt/xilinx/overlaybins/xdnnv3/xdnn_v3_96x16_2pe_8b_9mb_bank03.xclbin

Loading: '/opt/xilinx/overlaybins/xdnnv3/xdnn_v3_96x16_2pe_8b_9mb_bank03.xclbin'

-------------------

Speaking to Butler 

XRT build version: 2.3.1301

Build hash: 192e706aea53163a04c574f9b3fe9ed76b6ca471

Build date: 2019-10-24 20:05:16

Git branch: 2019.2

PID: 153

UID: 1001

[Thu Feb  6 02:35:37 2020]

HOST: zlabs-fpga

EXE: /opt/vitis_ai/conda/envs/vitis-ai-caffe/bin/python3.6

[XRT] ERROR: kernel 'pp_pipeline_accel' not found

./xf_pp_pipeline_tb.cpp:84 Error calling cl::Kernel krnl(program,kernelName,&errr), error code is: -46

Response from Butler is: 

errCode: errCode: 0

errCode String: SUCCESS

myHandle: 1

valid: 1

[XDNN] loading xclbin settings from /opt/xilinx/overlaybins/xdnnv3/xdnn_v3_96x16_2pe_8b_9mb_bank03.xclbin.json

[XDNN] using custom DDR banks 0,3

Path /tmp/vitis-ifpga/89/weights.h5 is a file.

Loading weights/bias/quant_params to FPGA...

[XRT]    git hash                   : 192e706aea53163a04c574f9b3fe9ed76b6ca471

[XDNN]   git hash                   : 1247bc4c889955a653bfefd1777c3aa2db3b4a4e

[XDNN] kernel configuration

[XDNN]   num cores                  : 2

[XDNN]   dsp array width            : 96

[XDNN]   axi data width (in 32bits) : 16

[XDNN]   img mem size               : 9 MB

[XDNN]   max instr num              : 1536

[XDNN]   max xbar entries           : 4096

[XDNN]   version                    : 3.2

[XDNN]   8-bit mode                 : 1
```

sumitn-xilinx, 2020-02-06:
Can you tell me the command you are running?
Because run.sh in https://github.com/Xilinx/Vitis-AI/tree/master/alveo/apps/xplusml/classification would take the xclbin from following:
/opt/xilinx/overlaybins/xdnnv3/xplusml/classification/

mahinlma, 2020-02-06:
in this directory  /opt/xilinx/overlaybins/xdnnv3/xplusml/classification/ , getting error - `xclbin not availble ,please build`
So I changed the xclbin directory to  /opt/xilinx/overlaybins/xdnnv3

sumitn-xilinx, 2020-02-06:
Please use this rpm to get xclbin:
https://www.xilinx.com/bin/public/openDownload?filename=xilinx-overlaybins-xplusml-1.0-0.x86_64.rpm
or 
https://www.xilinx.com/bin/public/openDownload?filename=xilinx-overlaybins-xplusml-18.04.deb

mahinlma, 2020-02-06:
@sumitn-xilinx, Thank you. 
Expected result will be preprocessing & classification response right.
I'm not getting classification results
Could you please let me know, what will be the issue.

```
`Running:

Test: test_classify

Model: googlenet_v1

Xclbin: /opt/xilinx/overlaybins/xdnnv3/xplusml/classification/

Accelerator: 0

python test_classify.py --xclbin /opt/xilinx/overlaybins/xdnnv3/xplusml/classification/ --netcfg ./data/googlenet_v1_8b_autoAllOpt.json --weights ./data/googlenet_v1_data.h5 --labels ./synset_words.txt --quantizecfg ./data/googlenet_v1_8b_xdnnv3.json --img_input_scale 1.0 --batch_sz 1 --vitis_rundir /tmp/vitis-ifpga/89 --images dog.jpg --deviceid 0

Found Platform

Platform Name: Xilinx

INFO: Reading /opt/xilinx/overlaybins/xdnnv3/xplusml/classification//xdnn_v3_96x16_2pe_8b_9mb_bank03.xclbin

Loading: '/opt/xilinx/overlaybins/xdnnv3/xplusml/classification//xdnn_v3_96x16_2pe_8b_9mb_bank03.xclbin'

-------------------

Speaking to Butler 

Pre-processing handle created. Populating Queue

XRT build version: 2.3.1301

Build hash: 192e706aea53163a04c574f9b3fe9ed76b6ca471

Build date: 2019-10-24 20:05:16

Git branch: 2019.2

PID: 153

UID: 1001

[Thu Feb  6 04:22:50 2020]

HOST: zlabs-fpga

EXE: /opt/vitis_ai/conda/envs/vitis-ai-caffe/bin/python3.6

[XRT] WARNING: unaligned host pointer '0x563ccad0a980' detected, this leads to extra memcpy

[XRT] WARNING: unaligned host pointer '0x7f36f35ff040' detected, this leads to extra memcpy

[XRT] WARNING: unaligned host pointer '0x7ffe6835b860' detected, this leads to extra memcpy

Response from Butler is: 

errCode: errCode: 0

errCode String: SUCCESS

myHandle: 1

valid: 1

Queue populated

[XDNN] loading xclbin settings from /opt/xilinx/overlaybins/xdnnv3/xplusml/classification/xdnn_v3_96x16_2pe_8b_9mb_bank03.xclbin.json

[XDNN] using custom DDR banks 0,3

Path /tmp/vitis-ifpga/89/weights.h5 is a file.

Loading weights/bias/quant_params to FPGA...

[XRT]    git hash                   : 192e706aea53163a04c574f9b3fe9ed76b6ca471

[XDNN]   git hash                   : 1247bc4c889955a653bfefd1777c3aa2db3b4a4e

[XDNN] kernel configuration

[XDNN]   num cores                  : 2

[XDNN]   dsp array width            : 96

[XDNN]   axi data width (in 32bits) : 16

[XDNN]   img mem size               : 9 MB

[XDNN]   max instr num              : 1536

[XDNN]   max xbar entries           : 4096

[XDNN]   version                    : 3.2

[XDNN]   8-bit mode                 : 1
```

`

mahinlma, 2020-02-06:
@sumitn-xilinx , Thank you.
able run x+ml application.

mahinlma, 2020-02-06:
@sumitn-xilinx , please tell me, the reason behind these accuracy loss

googlenet_v1 accuracy
```
---------- Prediction 1/1 for /workspace/alveo/apps/xplusml/classification/dog.jpg ----------

0.0021 "n04005630 prison, prison house"

0.0021 "n03000134 chainlink fence"

0.0021 "n03291819 envelope"

0.0019 "n04350905 suit, suit of clothes"

0.0019 "n02786058 Band Aid"
```

resnet50 - accuracy
```

---------- Prediction 1/1 for /workspace/alveo/apps/xplusml/classification/dog.jpg ----------

0.0010 "n03657121 lens cap, lens cover"

0.0010 "n07615774 ice lolly, lolly, lollipop, popsicle"

0.0010 "n04356056 sunglasses, dark glasses, shades"

0.0010 "n04597913 wooden spoon"

0.0010 "n03759954 microphone, mike"

```

nkpavan, 2020-02-07:
@mahinlma , 
Please reset the card using command `/opt/xilinx/xrt/bin/xbutil reset` and make sure that Googlenet and resnet-50 with pre-processing being run on the host is functional. You can run that by going to directory /workspace/alveo/examples/deployment_modes and execute run.sh. If it's functional, go to directory /workspace/alveo/apps/xplusml/classification/ and execute run.sh

mahinlma, 2020-02-08:
Thank you @nkpavan

closed 2020-02-08 by mahinlma

sina1989, 2020-04-08:
Hi

I see an accuracy loss when i want to run vitis ai samples for Alveo in

https://github.com/Xilinx/Vitis-AI/tree/master/alveo/examples/vitis_ai_alveo_samples

i followed instructions in

https://github.com/Xilinx/Vitis-AI/issues/41

i changed default xilinx_u200_xdma_201830_2 in instal.sh to  xilinx_u200_xdma_201830_1 and flashed my Alveo U200 card again. i reset card by "/opt/xilinx/xrt/bin xbutil reset" command but nothing solved. for example in deployment_modes example i get following report:

 

Using VAI_ALVEO_ROOT
------------------
/workspace/alveo

---------------------
Using LD_LIBRARY_PATH
---------------------

-------------------
Using LIBXDNN_PATH
-------------------
/opt/vitis_ai/conda/envs/vitis-ai-caffe/lib/libxfdnn.so

-------------------
PYTHONPATH
-------------------
/opt/vitis_ai/compiler

---------------------
Verifying XILINX_XRT
---------------------
XILINX_XRT : /opt/xilinx/xrt
PATH : /opt/xilinx/xrt/bin:/opt/vitis_ai/conda/envs/vitis-ai-caffe/bin:/opt/vitis_ai/conda/condabin:/opt/vitis_ai/conda/bin:/opt/vitis_ai/utility:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
LD_LIBRARY_PATH : /opt/xilinx/xrt/lib:
PYTHONPATH : /opt/xilinx/xrt/python:/opt/vitis_ai/compiler
--- Using System XCLBIN ---
Running:
Test: test_classify
Model: googlenet_v1
Xclbin: /opt/xilinx/overlaybins/xdnnv3
Accelerator: 0

python test_classify.py --xclbin /opt/xilinx/overlaybins/xdnnv3 --netcfg ./data/googlenet_v1_8b_autoAllOpt.json --weights ./data/googlenet_v1_data.h5 --labels ./synset_words.txt --quantizecfg ./data/googlenet_v1_8b_xdnnv3.json --img_input_scale 1.0 --batch_sz 4 --vitis_rundir /tmp/vitis-sahand/786 --images dog.jpg
-------------------
Speaking to Butler
Response from Butler is:
errCode: errCode: 0
errCode String: SUCCESS
myHandle: 1
valid: 1

[XDNN] loading xclbin settings from /opt/xilinx/overlaybins/xdnnv3/xdnn_v3_96x16_2pe_8b_9mb_bank03.xclbin.json
[XDNN] using custom DDR banks 0,3
Path /tmp/vitis-sahand/786/weights.h5 is a file.
Loading weights/bias/quant_params to FPGA...

[XRT] git hash : 192e706aea53163a04c574f9b3fe9ed76b6ca471
[XDNN] git hash : 1247bc4c889955a653bfefd1777c3aa2db3b4a4e
[XDNN] kernel configuration
[XDNN] num cores : 2
[XDNN] dsp array width : 96
[XDNN] axi data width (in 32bits) : 16
[XDNN] img mem size : 9 MB
[XDNN] max instr num : 1536
[XDNN] max xbar entries : 4096
[XDNN] version : 3.2
[XDNN] 8-bit mode : 1
---------- Prediction 1/4 for /workspace/alveo/examples/deployment_modes/dog.jpg ----------
0.0021 "n04005630 prison, prison house"
0.0021 "n03000134 chainlink fence"
0.0021 "n03291819 envelope"
0.0019 "n04350905 suit, suit of clothes"
0.0019 "n02786058 Band Aid"