Xilinx/Vitis-AI #42: xclbinutil is not found in path

beldachi, 2020-02-07:
Hi
I am following the tutorials available at https://github.com/Xilinx/Vitis-AI/blob/master/DPU-TRD/prj/Vitis/README.md.In the section "5.2.1 Building the hardware Design", when I  run the "% make KERNEL=DPU_SM DEVICE=zcu102" to build the hardware, I can see the bit file is generated, but I receive the following error after that: 

ERROR: [v++ 82-29] Exception in main() -- ERROR: [v++ 17-70] Application Exception: Not found in path: xclbinutil
Makefile:70: recipe for target 'binary_container_1/dpu.xclbin' failed
 
Here is the terminal screen for the related error:

[19:11:05] Starting bitstream generation..
[19:13:06] Creating bitmap...
[19:13:37] Writing bitstream ./zcu102_base_wrapper.bit...
[19:13:37] Finished 6th of 6 tasks (FPGA bitstream generation). Elapsed time: 00h 02m 31s 
[19:13:37] Run vpl: Step impl: Completed
[19:13:38] Run vpl: FINISHED. Run Status: impl Complete!
INFO: [v++ 60-1441] [19:13:38] Run run_link: Step vpl: Completed
Time (s): cpu = 00:36:26 ; elapsed = 00:49:59 . Memory (MB): peak = 684.215 ; gain = 0.000 ; free physical = 54560 ; free virtual = 63819
INFO: [v++ 60-1443] [19:13:38] Run run_link: Step rtdgen: Started
INFO: [v++ 60-1453] Command Line: rtdgen
INFO: [v++ 60-1454] Run Directory: /opt/AI/Vitis-AI/DPU-TRD/prj/Vitis/binary_container_1/link/run_link
INFO: [v++ 60-1453] Command Line: cf2sw -a /opt/AI/Vitis-AI/DPU-TRD/prj/Vitis/binary_container_1/link/int/address_map.xml -sdsl /opt/AI/Vitis-AI/DPU-TRD/prj/Vitis/binary_container_1/link/int/sdsl.dat -xclbin /opt/AI/Vitis-AI/DPU-TRD/prj/Vitis/binary_container_1/link/int/xclbin_orig.xml -rtd /opt/AI/Vitis-AI/DPU-TRD/prj/Vitis/binary_container_1/link/int/dpu.rtd -o /opt/AI/Vitis-AI/DPU-TRD/prj/Vitis/binary_container_1/link/int/dpu.xml
INFO: [v++ 60-1618] Launching 
INFO: [v++ 60-1441] [19:13:40] Run run_link: Step rtdgen: Completed
Time (s): cpu = 00:00:02 ; elapsed = 00:00:02 . Memory (MB): peak = 684.215 ; gain = 0.000 ; free physical = 54561 ; free virtual = 63821
INFO: [v++ 60-1443] [19:13:40] Run run_link: Step xclbinutil: Started
INFO: [v++ 60-1453] Command Line: xclbinutil --add-section BITSTREAM:RAW:/opt/AI/Vitis-AI/DPU-TRD/prj/Vitis/binary_container_1/link/int/system.bit --force --key-value SYS:mode:flat --add-section :JSON:/opt/AI/Vitis-AI/DPU-TRD/prj/Vitis/binary_container_1/link/int/dpu.rtd --add-section CLOCK_FREQ_TOPOLOGY:JSON:/opt/AI/Vitis-AI/DPU-TRD/prj/Vitis/binary_container_1/link/int/dpu_xml.rtd --add-section BUILD_METADATA:JSON:/opt/AI/Vitis-AI/DPU-TRD/prj/Vitis/binary_container_1/link/int/dpu_build.rtd --add-section EMBEDDED_METADATA:RAW:/opt/AI/Vitis-AI/DPU-TRD/prj/Vitis/binary_container_1/link/int/dpu.xml --add-section SYSTEM_METADATA:RAW:/opt/AI/Vitis-AI/DPU-TRD/prj/Vitis/binary_container_1/link/int/systemDiagramModelSlrBaseAddress.json --key-value SYS:PlatformVBNV:xilinx_zcu102_zcu102_base_1_0 --output /opt/AI/Vitis-AI/DPU-TRD/prj/Vitis/binary_container_1/link/int/dpu.xclbin
INFO: [v++ 60-1454] Run Directory: /opt/AI/Vitis-AI/DPU-TRD/prj/Vitis/binary_container_1/link/run_link
ERROR: [v++ 82-29] Exception in main() -- ERROR: [v++ 17-70] Application Exception: Not found in path: xclbinutil
 
Makefile:70: recipe for target 'binary_container_1/dpu.xclbin' failed
make: *** [binary_container_1/dpu.xclbin] Error 1
root@hpn118:/opt/AI/Vitis-AI/DPU-TRD/prj/Vitis# 
 

Could you please advise me how to resolve this issue?

I would greatly appreciate if someone answer my query.

Thank you.

jimheaton, 2020-02-13:
Did you source the following before starting the run:
source $XILINX_XRT/setup.sh
Where $XILINX_XRT points to your xrt install, for example /opt/xilinx/xrt

I saw the exact same error, but then after sourcing this then I could find xclbinutil

EXTERNAL EMAIL

beldachi, 2020-02-13:
Hi ;
I solved the issue by installing fresh Ubuntu 18.04. The previous one was installed a couple of years ago and somehow caused this error and other problems.

closed 2020-02-13 by beldachi