Xilinx/Vitis-AI #44: Is it possible to use bit/activation_width = 16 for quantizing tensorflow models?

katherineyun, 2020-02-21:
Hi,

I am trying to quantize a tensorflow model with bit and activation widths set to 16 instead of 8, hoping to increase the final top-1/top-5 accuracies. However, when I tried this by calling vai_q_tensorflow and included --weight_bit 16 --activation_bit 16, the final accuracies were close to 0. I was wondering why this happened?

Any help is appreciated.

Thanks,

jimheaton, 2020-03-09:
For deployment on the DPU only 8 bits are allowed. You can use 16 or 32 bits to check quantizer accuracy only but you wont be able to deploy on HW>

If you did not see an accuracy increase out of quantizer. There are a couple of things to check. First check that you are using at least 100 calibration images. You can also try using a different value for the method option. If you are using 1(default), try 0.

qianglin-xlnx, 2020-05-11:
Hi @kamranjk 
As suggested by @jimheaton , I wonder if you have solved this issue. If not, please describe your steps in detail and I will submit this issue to the engineer in charge of it. If it has been resolved, please close this issue.
Note that if we do not receive any reply from you within 2 weeks, we will assume that this issue has been resolved and will close this issue. Thank you.

closed 2020-05-11 by katherineyun