Xilinx/Vitis-AI #45: Running vitis-ai zcu104 examples results in black screen

mullerrwd, 2020-02-24:
Running any of the examples from `vitis_ai_samples_zcu104` below results in a black screen and screen entering sleep mode. 

- `adas_detection`
- `resnet50`
- `resnet50_mt_py`
- `inception_v1`
- `video_analysis`
- `segmentation`

The `pose_detection` example however completes successfully and the `video_analysis` example runs for a few seconds before the screen enters sleep mode as well. 

**All examples were tested straight from ZCU104 terminal, not over ssh with X11 forwarding!**

## Steps to Reproduce
1. From within the Vitis-AI `runtime` docker container, cross-compile all `vitis_ai_samples_zcu104`  and transfer to board. Including any necessary test images in a folder `dataset` within each example dir as outlined in [UG1414, page 28].(https://www.xilinx.com/support/documentation/sw_manuals/vitis_ai/1_0/ug1414-vitis-ai.pdf)
2. Connect DP capable screen and peripherals (usb-hub, mouse, keyboard)
3. Run one of the examples outlined above:
  a. `./resnet50 dpuv2_rundir`
  b. `python3 resnet50.py {thread_number} dpuv2_rundir`
  c. `python3 inception_v1.py {thread_number} dpuv2_rundir`
 d. `etc` 
4. Screen looses connection instantly and enters sleep mode.

No logs are available in `/var/log/` apart from boot logs. Tried different monitors and display port cables. 

## Specifications
- Version: Vitis-AI Release 1.0 936881527ed9a4dcd4956b1269ab157c888f4be6
- Platform: ZCU104 (HW-Z1-ZCU104 Rev1.0)
- Board Image: [xilinx-zcu104-dpu-v2019.2img.gz](https://www.xilinx.com/bin/public/openDownload?filename=xilinx-zcu104-dpu-v2019.2.img.gz)

## Related issues
Seem related to issue #31 described by @Morett07.

jimheaton, 2020-02-24:
Since you are running from a serial terminal have you set following envar?
export DISPLAY=:0.0

EXTERNAL EMAIL

mullerrwd, 2020-02-26:
I might have been unclear. No, I'm not running the examples from a serial connection but rather directly from terminal on the ZCU104 standalone. 

However, I tried several examples over serial connection. And your suggestion to set `DISPLAY` envar to `:0.0`. Default is `:0`. I have included the output below. In all cases, serial terminal is unresponsive after screen failure. I then have to reboot the ZCU104. 

---

Output of `adas_detection` example over serial:
```bash
root@xilinx-zcu104-2019_2:~/vitis_ai_samples_zcu104/adas_detection# export DISPLAY=:0.0
root@xilinx-zcu104-2019_2:~/vitis_ai_samples_zcu104/adas_detection# ./adas_detection video/adas.avi dpuv2_rundir/
[  281.833992] [drm] Pid 2669 opened device
[  281.837942] [drm] Pid 2669 closed device
[  281.842038] [drm] Pid 2669 opened device
[  281.873837] [drm] The XCLBIN already loaded. Don't need to reload.
[  281.876067] [drm] Reconfiguration not supported
[  281.891668] [drm] Pid 2669 opened device
[  281.896093] [drm] CU can only be initialized once.
[  282.039276] [drm] Pid 2669 opened device
[  282.064790] [drm] Pid 2669 opened device
[  282.095679] [drm] Pid 2669 opened device

** (ADAS Detection@Xlinx DPU:2669): CRITICAL **: 19:13:15.208: Failed to get connection to xfconfd: Could not connect: Connection refused
```
The screen goes black instantly after the CRITICAL error. Serial connection is unresponsive.

---

Output of `video_analysis`
```bash
root@xilinx-zcu104-2019_2:~/vitis_ai_samples_zcu104/video_analysis# export DISPLAY=:0.0
root@xilinx-zcu104-2019_2:~/vitis_ai_samples_zcu104/video_analysis# ./video_analysis video/structure.mp4 dpuv2_rundir/
Detect video: video/structure.mp4
[  118.584927] [drm] Pid 2673 opened device
[  118.588888] [drm] Pid 2673 closed device
[  118.592888] [drm] Pid 2673 opened device
[  118.624685] [drm] The XCLBIN already loaded. Don't need to reload.
[  118.626922] [drm] Reconfiguration not supported
[  118.642603] [drm] Pid 2673 opened device
[  118.647010] [drm] CU can only be initialized once.
[  119.012714] [drm] Pid 2673 opened device
[  119.021452] [drm] Pid 2673 opened device
[  119.043873] [drm] Pid 2673 opened device

** (Video Analysis @Xilinx DPU:2673): CRITICAL **: 19:18:04.911: Failed to get connection to xfconfd: Could not connect: Connection refused
```
The screen goes black several seconds after the CRITICAL error and serial is unresponsive. Thus I do see the example running for several seconds before the screen failure as experienced when running example directly from terminal. 

---

Output of `segmentation` example.
```bash
root@xilinx-zcu104-2019_2:~/vitis_ai_samples_zcu104/segmentation# export DISPLAY=:0.0
root@xilinx-zcu104-2019_2:~/vitis_ai_samples_zcu104/segmentation# ./segmentation
Makefile      dpuv2_rundir/ segmentation  video/
build/        model/        src/
root@xilinx-zcu104-2019_2:~/vitis_ai_samples_zcu104/segmentation# ./segmentation
 video/traffic.mp4 dpuv2_rundir/
Detect video: video/traffic.mp4
[   60.091567] [drm] Pid 2643 opened device
[   60.095524] [drm] Pid 2643 closed device
[   60.099508] [drm] Pid 2643 opened device
[   62.840751] [drm] Finding IP_LAYOUT section header
[   62.840759] [drm] Section IP_LAYOUT details:
[   62.845560] [drm]   offset = 0x126ae48
[   62.849823] [drm]   size = 0xa8
[   62.853704] [drm] Finding DEBUG_IP_LAYOUT section header
[   62.856851] [drm] AXLF section DEBUG_IP_LAYOUT header not found
[   62.862161] [drm] Finding CONNECTIVITY section header
[   62.868077] [drm] Section CONNECTIVITY details:
[   62.873127] [drm]   offset = 0x126aef0
[   62.877647] [drm]   size = 0xf4
[   62.881395] [drm] Finding MEM_TOPOLOGY section header
[   62.884532] [drm] Section MEM_TOPOLOGY details:
[   62.889581] [drm]   offset = 0x126ad50
[   62.894104] [drm]   size = 0xf8
[   62.899896] [drm] No ERT scheduler on MPSoC, using KDS
[   62.908600] [drm] scheduler config ert(0)
[   62.908603] [drm]   cus(2)
[   62.912611] [drm]   slots(16)
[   62.915309] [drm]   num_cu_masks(1)
[   62.918281] [drm]   cu_shift(16)
[   62.921752] [drm]   cu_base(0x80000000)
[   62.924974] [drm]   polling(0)
[   62.933632] [drm] Pid 2643 opened device
[   63.974111] [drm] Pid 2643 opened device
[   63.978250] [drm] Pid 2643 opened device
```
Screen goes black after last message above. No `xconfd` error however?

---

The `resnet` example.
```bash
root@xilinx-zcu104-2019_2:~/vitis_ai_samples_zcu104/resnet50# export DISPLAY=:0.0
root@xilinx-zcu104-2019_2:~/vitis_ai_samples_zcu104/resnet50# ./resnet50 dpuv2_rundir/
[   47.399439] [drm] Pid 2636 opened device
[   47.403381] [drm] Pid 2636 closed device
[   47.407378] [drm] Pid 2636 opened device
[   49.992421] [drm] Finding IP_LAYOUT section header
[   49.992429] [drm] Section IP_LAYOUT details:
[   49.997252] [drm]   offset = 0x126ae48
[   50.001520] [drm]   size = 0xa8
[   50.005285] [drm] Finding DEBUG_IP_LAYOUT section header
[   50.008423] [drm] AXLF section DEBUG_IP_LAYOUT header not found
[   50.013740] [drm] Finding CONNECTIVITY section header
[   50.019650] [drm] Section CONNECTIVITY details:
[   50.024692] [drm]   offset = 0x126aef0
[   50.029225] [drm]   size = 0xf4
[   50.032985] [drm] Finding MEM_TOPOLOGY section header
[   50.036126] [drm] Section MEM_TOPOLOGY details:
[   50.041167] [drm]   offset = 0x126ad50
[   50.045695] [drm]   size = 0xf8
[   50.051812] [drm] No ERT scheduler on MPSoC, using KDS
[   50.060515] [drm] scheduler config ert(0)
[   50.060518] [drm]   cus(2)
[   50.064528] [drm]   slots(16)
[   50.067233] [drm]   num_cu_masks(1)
[   50.070192] [drm]   cu_shift(16)
[   50.073665] [drm]   cu_base(0x80000000)
[   50.076884] [drm]   polling(0)
[   50.085650] [drm] Pid 2636 opened device
[   53.937630] [drm] Pid 2636 opened device
[   53.941563] [drm] Pid 2636 opened device
```
Again, screen goes black after last message and serial is unresponsive.

jimheaton, 2020-02-27:
I am wondering if the issue is with the display. Have you tried running the display over ssh?

mullerrwd, 2020-02-28:
Unfortunately, almost same behavior trying over ssh with `X11 forwarding`. 

Some notes:
The `pose_detection` example completed successfully as expected. 
The `video_analysis` ran longer than expected. 
Furthermore, the `segmentation` and `adas_detection` examples show at most the first frame, given the example is started after a fresh reboot of the `ZCU104` board. 
The other examples are basically non-responding.

qianglin-xlnx, 2020-05-11:
hi Mullerrwd,
I wonder if you have solved this problem. If not, I will submit this issue to the engineer in charge of it. If it has been resolved, please let me know and I will close this issue. 
Note that if we do not receive any reply from you within 2 weeks, we will assume that this issue has been resolved and will close this issue. Thank you.

qianglin-xlnx, 2020-05-11:
Note: You may need to run command irps5401 to trigger the power management patch for ZCU104 to avoid system hang or power off issue when running Vitis™ AI examples.

mullerrwd, 2020-05-25:
> Note: You may need to run command irps5401 to trigger the power management patch for ZCU104 to avoid system hang or power off issue when running Vitis™ AI examples.

Hi @qianglin-xlnx thank you for your reply, I'll make sure to test the power management patch tonight. I'll report back afterwards. Thanks

zackdilan, 2020-05-27:
@qianglin-xlnx I am having DPU timeout error while running the vistis AI examples. Can you elaborate on how to run this irps5401 command?
TA

qianglin-xlnx, 2020-05-27:
@zackdilan It's a binary executable that can be found in the following image.
https://www.xilinx.com/bin/public/openDownload?filename=xilinx-zcu104-dpu-v2019.2-v2.img.gz

Just execute it as shown in the following screenshot.

[image]

zackdilan, 2020-05-27:
@qianglin-xlnx thanks I couldn't find it in the image I have **xilinx-zcu104-dpu-v2019.2.img.gz**. I will try with the image that you have provided.

zackdilan, 2020-05-27:
okay now I can find it with the new image, But now I haveing some issue while installing the Vitis AI library.
[image]

mullerrwd, 2020-05-27:
@zackdilan Hi Zackdilan. Like me, you are probably using the previous prebuilt `ZCU10X` [image](https://www.xilinx.com/bin/public/openDownload?filename=xilinx-zcu104-dpu-v2019.2.img.gz). The `irps5401` seems unavailable in the older image. The updated documentation however is not clear on this as there is a new `V2` prebuilt image available for both edge devices. 

After flashing and installing the [VART](https://github.com/Xilinx/Vitis-AI/blob/master/VART/README.md#quick-start-for-edge) to the evaluation board the `irps5401` command should be available.

mullerrwd, 2020-05-27:
Applying the power management patch within the new prebuilt `ZCU10X` image solved the problems described in this issue. Closed!

closed 2020-05-27 by mullerrwd