Xilinx/Vitis-AI #46: Using WideType Data Container to Increase Memory Bit-Width breaks Multi-CU support

dylanbespalko, 2020-02-25:
## Problem
I am adding complex number support to[ PyTorch on the FPGA](https://gitlab.com/pytorch-complex/vitis_kernels) and I rely heavily on the ability to pass multiple samples to the FPGA at the same time using your [WideType](https://github.com/Xilinx/Vitis_Libraries/blob/40b98f7a20d4f9d181642104f74d797a94315024/blas/L1/include/hw/xf_blas/helpers/utils/types.hpp#L41) Single-Instruction, Multiple Data (SIMD) container. For a 512-bit memory bus, I would like to stream the following using my SIMD [Vec<T, LOG2BW>](https://gitlab.com/pytorch-complex/vitis_kernels/-/blob/master/aten/src/ATen/fpga/vec/vec_base.h) Container:

real float:  [R0, R1, R2, R3, R4, R5, R6, R7, R8, R9, R10, R11, R12, R13, R14, R15]
complex float[2]:  [R0, I0, R1, I1, R2, I2, R3, I3, R4, I4, R5, I5, R6, I6, R7, I7]
quaternion float[4]:  [R0, I0, J0, K0, R1, I1, J1, K1, R2, I2, J2, K2, R3, I3, J3, K3]

## Background
The complex/quaternion support is needed for RF/Optical communications. I have seen a wide variety of complex number support in the HLS documentations

```
|      Type                       |     Synthesis             |          Simulation        |
--------------------------------------------------------------------------------------------
|      std::complex <ap_fixed <>> |             YES           |              YES           |
|      std::complex<float>        |              NO           |              YES           |
|      std::complex<double>       |              NO           |              YES           |
|      complex_wrapper<float>     |              NO           |              YES           |
|      complex_wrapper<double>    |              YES          |              YES           |
```

The first three support the `R, I, R, I` memory format, while the last 2 support `R, R, I, I` memory format.

## Proposed Solution

Now that Xilinx Vitis allows us to use Vectorized SIMD container datatypes (see [WideType](https://github.com/Xilinx/Vitis_Libraries/blob/40b98f7a20d4f9d181642104f74d797a94315024/blas/L1/include/hw/xf_blas/helpers/utils/types.hpp#L41)), there should be a built in way to treat `std::complex` data as a `float[2]` and `std::quaternion` as `float[4]`. If this idea works, the PL code wouldn't need to support `std::complex<T>` containers (which it currently doesn't).

I have written a blog about this: 
[pytorch-for-fpga-part-1-heterogeneous-processing](https://www.dylan-bespalko.scikit-nonlinear.org/post/pytorch-for-fpga-part-1-heterogeneous-processing)
[pytorch-for-fpga-part-2-basic-fpga-optimizations](https://www.dylan-bespalko.scikit-nonlinear.org/post/pytorch-for-fpga-part-2-basic-fpga-optimizations)
[pytorch-for-fpga-part-3-advanced-fpga-optimizations](https://www.dylan-bespalko.scikit-nonlinear.org/post/pytorch-for-fpga-part-3-advanced-fpga-optimizations)
[pytorch-for-fpga-part-4-deploying-pytorch-kernels](https://www.dylan-bespalko.scikit-nonlinear.org/post/pytorch-for-fpga-part-4-deploying-pytorch-kernels)
[pytorch-for-fpga-part-5-complex-and-quaternion-numbers](https://www.google.com)(Coming Soon)

In Part-4 of this blog, SIMD has weird profile performance and it doesn't compile at all when the number of CUs is greater than 1. Here are some questions:

1. Does the hw_emu of WideType predict the kernel duration time accurately (see Detail)? Can you provide an Example?
2. Does the WideType data container work with Multiple Compute Units (CU)? Can you provide an Example?
3. Do you see any long-term problems with supporting complex numbers in this way? Are there potential Bottlenecks? I'm still pretty new to this.

## Detail

Here are some logs. Vec<float, 0> (2^0 = 1) achieves the same hw_emu speed as float, which is expected. The Iteration intervals make sense. I'm not sure how to calculate the depth, but the **duration** of these math kernels seems to go up and down arbitrarily.

```
float duration 0.063ms
INFO: [v++ 204-61] Pipelining loop 'read'.
INFO: [v++ 204-61] Pipelining result : Target II = 1, Final II = 1, Depth = 3.
INFO: [v++ 204-61] Pipelining loop 'vadd'.
INFO: [v++ 204-61] Pipelining result : Target II = 1, Final II = 1, Depth = 14.
INFO: [v++ 204-61] Pipelining loop 'write'.
INFO: [v++ 204-61] Pipelining result : Target II = 1, Final II = 1, Depth = 3.
INFO: [v++ 200-789] **** Estimated Fmax: 411.02 MHz

Vec<float, 0>  duration 0.062ms
INFO: [v++ 204-61] Option 'relax_ii_for_timing' is enabled, will increase II to preserve clock frequency constraints.
INFO: [v++ 204-61] Pipelining loop 'read'.
INFO: [v++ 204-61] Pipelining result : Target II = 1, Final II = 1, Depth = 3.
INFO: [v++ 204-61] Pipelining loop 'vadd'.
INFO: [v++ 204-61] Pipelining result : Target II = 1, Final II = 1, Depth = 14.
INFO: [v++ 204-61] Pipelining loop 'write'.
INFO: [v++ 204-61] Pipelining result : Target II = 1, Final II = 1, Depth = 3.
INFO: [v++ 200-789] **** Estimated Fmax: 411.02 MHz

Vec<float, 1> duration 0.204ms
INFO: [v++ 204-61] Pipelining loop 'read'.
INFO: [v++ 204-61] Pipelining result : Target II = 2, Final II = 2, Depth = 11.
INFO: [v++ 204-61] Pipelining loop 'vadd'.
INFO: [v++ 204-61] Pipelining result : Target II = 2, Final II = 2, Depth = 15.
INFO: [v++ 204-61] Pipelining loop 'write'.
INFO: [v++ 204-61] Pipelining result : Target II = 2, Final II = 2, Depth = 9

Vec<float, 2> duration 0.178ms
INFO: [v++ 204-61] Pipelining result : Target II = 4, Final II = 4, Depth = 13.
INFO: [v++ 204-61] Pipelining loop 'vadd'.
INFO: [v++ 204-61] Pipelining result : Target II = 4, Final II = 4, Depth = 17.
INFO: [v++ 204-61] Pipelining loop 'write'.
INFO: [v++ 204-61] Pipelining result : Target II = 4, Final II = 4, Depth = 11.
INFO: [v++ 200-789] **** Estimated Fmax: 411.02 MHz

Vec<float, 3> duration 0.205ms
INFO: [v++ 204-61] Pipelining result : Target II = 8, Final II = 8, Depth = 17.
INFO: [v++ 204-61] Pipelining loop 'vadd'.
INFO: [v++ 204-61] Pipelining result : Target II = 8, Final II = 8, Depth = 21.
INFO: [v++ 204-61] Pipelining loop 'write'.
INFO: [v++ 204-61] Pipelining result : Target II = 8, Final II = 8, Depth = 15.
INFO: [v++ 200-789] **** Estimated Fmax: 411.02 MHz

Vec<float, 4> duration 0.144ms
INFO: [v++ 204-61] Pipelining loop 'read'.
INFO: [v++ 204-61] Pipelining result : Target II = 16, Final II = 16, Depth = 25.
INFO: [v++ 204-61] Pipelining loop 'vadd'.
INFO: [v++ 204-61] Pipelining result : Target II = 16, Final II = 16, Depth = 29.
INFO: [v++ 204-61] Pipelining loop 'write'.
INFO: [v++ 204-61] Pipelining result : Target II = 16, Final II = 16, Depth = 23.
```

anjn, 2020-02-26:
@dylanbespalko Great articles!

When you increase the bus width, usually you increase the number of arithmetic units accordingly. Seeing the log, `Vec<float, 4>` achieves II=16. This means the IO bandwidth becomes 16x, but the throughput is the same as the `Vec<float, 0>` case. II should be always 1 for all the data types.

I suggest modifying your code as follows.

[vec_base.h](https://gitlab.com/pytorch-complex/vitis_kernels/-/blob/2c0ffbae1f96e8bfad5f356d2ffed691501ef3f4/aten/src/ATen/fpga/vec/vec_base.h#L217)
```cpp
  Vec<T, LOG2BW> operator+(const Vec<T, LOG2BW>& other) const {
#pragma HLS inline // anjn: 'self' is not necessary
    using VH = VecHost<T, LOG2BW>;
    Vec<T, LOG2BW> ret;
    for (size_t i = 0; i < size(); i++) {
#pragma HLS UNROLL // anjn: Duplicate arthmetic unit
      ret[i] = VH::values[i] + other[i];
    }
    return ret;
  }
```
[vadd.hpp](https://gitlab.com/pytorch-complex/vitis_kernels/-/blob/2c0ffbae1f96e8bfad5f356d2ffed691501ef3f4/aten/src/ATen/fpga/kernel/BinaryArithmeticKernel/vadd.hpp#L4)
```
template <typename DTYPE>
void vadd_dataflow(hls::stream<DTYPE>& result_stream,
               hls::stream<DTYPE>& a_stream,
               hls::stream<DTYPE>& b_stream,
               DTYPE alpha,
               size_t stride) {
  DTYPE a, b;
  vadd: for (size_t j = 0; j < stride*DTYPE::rn()/DTYPE::size(); j++){
#pragma HLS PIPELINE II=1 // anjn: II should be 1
    a_stream >> a;
    b_stream >> b;
    result_stream << a + b;
  }
}
```

`Dataflow.h` is also needed to be modified so that the read/write functions also achieve II=1.

For the measured duration,
```
float duration 0.063ms
Vec<float, 0>  duration 0.062ms
Vec<float, 1> duration 0.204ms
Vec<float, 2> duration 0.178ms
Vec<float, 3> duration 0.205ms
Vec<float, 4> duration 0.144ms
```
I'm not sure how you measured the duration, I guess this includes various overhead other than kernel execution time, for example, data movement, kernel invocation, interrupt to host. The simplest way to hide these overheads is by using larger data.

dylanbespalko, 2020-02-26:
@anjn,

Thanks for your quick reply and I think I get what you mean. I am gone until Thursday. I will try to fix it and close the issue then.

dylanbespalko, 2020-03-01:
@anjn,

Thank you so much for your help. Your suggestions reduced the II of `vadd_dataflow()` to 1.  I have also been trying to fix the read/write functions in `Dataflow.h` with no success.

I have attached a simplified example `WideTypeMemoryRead.txt (.cpp)`  where most of the code is copied directly from the Vitis Libraries.

Line6: class BitConv is from `blas/L1/include/hw/xf_blas/helpers/utils/types.hpp`
Line37: class WideType is from `blas/L1/include/hw/xf_blas/helpers/utils/types.hpp`
Line103: void readVec2Stream is from `blas/L1/include/hw/xf_blas/helpers/dataMover/vecMoverB1.hpp`

The only original code in this file is:
```C++
extern "C" {

void vfadd(float a[2]) {
#pragma HLS INTERFACE s_axilite port=a  bundle=control
#pragma HLS INTERFACE s_axilite port=return bundle=control

#pragma HLS INTERFACE m_axi port=a  offset=slave bundle=gmem2

hls::stream<WideType<float, 2> > l_strX;
#pragma HLS DATAFLOW
#pragma HLS data_pack variable=a
#pragma HLS data_pack variable = l_strX
//#pragma HLS ARRAY_PARTITION variable=a dim=0 complete
//#pragma HLS ARRAY_PARTITION variable=l_strX dim=0 complete
//#pragma HLS ARRAY_PARTITION variable=a cyclic factor=2 partition
//#pragma HLS ARRAY_PARTITION variable=l_strX cyclic factor=2 partition
  readVec2Stream<float, 2>(a, 2, l_strX);
}

}
```

I have tried every pragma I can think of to widen the data path, but I keep getting the following compiler response:
```bash
===>The following messages were generated while  performing high-level synthesis for kernel: vfadd Log file: /home/jenkins/repos/vitis_kernels/dev/BinaryOpsKernel/v_add/build/kernel/_x/vfadd/vfadd/vivado_hls.log :
INFO: [v++ 204-61] Option 'relax_ii_for_timing' is enabled, will increase II to preserve clock frequency constraints.
INFO: [v++ 204-61] Pipelining function 'readVec2Stream<float, 2u>'.
WARNING: [v++ 204-69] Unable to schedule bus request on port 'p_in_0' (/home/jenkins/repos/vitis_kernels/dev/BinaryOpsKernel/v_add/kernel/vec/vfadd/vfadd.cpp:115) due to limited memory ports. Please consider using a memory core with more ports or partitioning the array.
INFO: [v++ 204-61] Pipelining result : Target II = 1, Final II = 2, Depth = 9.
INFO: [v++ 200-789] **** Estimated Fmax: 291.29 MHz
INFO: [v++ 60-594] Finished kernel compilation
INFO: [v++ 60-244] Generating system estimate report...
INFO: [v++ 60-1092] Generated system estimate report: /home/jenkins/repos/vitis_kernels/dev/BinaryOpsKernel/v_add/build/vfadd/system_estimate_vfadd.xtxt
Add Instance readVec2Stream_float_2u_s readVec2Stream_float_2u_U0 66
INFO: [v++ 60-586] Created /home/jenkins/repos/vitis_kernels/dev/BinaryOpsKernel/v_add/build/kernel/vfadd/vfadd.xo
INFO: [v++ 60-791] Total elapsed time: 0h 0m 35s
```

The warning is telling me to partition the array, but nothing I have tried seems to achieve II=1. How to I partition the array so that the `readVec2Stream` achieves an `II=1`?

[WideTypeMemoryRead.txt](https://github.com/Xilinx/Vitis-AI/files/4272258/WideTypeMemoryRead.txt)

anjn, 2020-03-02:
@dylanbespalko,

You are trying to fill `WideType<float,2>` with a 32-bit interface, so it takes 2 cycles. Your code should look like this.

```cpp
extern "C" {

void vfadd(ap_uint<64>* a, int length) {
#pragma HLS INTERFACE s_axilite port=a  bundle=control
#pragma HLS INTERFACE s_axilite port=length  bundle=control
#pragma HLS INTERFACE s_axilite port=return bundle=control
#pragma HLS INTERFACE m_axi port=a  offset=slave bundle=gmem2

#pragma HLS DATAFLOW

  hls::stream<WideType<float, 2> > l_strX;
#pragma HLS data_pack variable = l_strX

  //readVec2Stream<float, 2>(a, 2, l_strX);
  for (int i=0; i<length; i++) {
#pragma HLS PIPELINE II=1
    WideType<float, 2> f_vec;
    ap_uint<64> a_dat = a[i];
    for (int j=0; j<2; j++) {
      BitConv<float> conv_f;
      f_vec[j] = conv_f.toType(a_dat(32*j+31,32*j));
    }
    l_strX << f_vec;
  }
}

}
```

dylanbespalko, 2020-03-15:
@anjn,

Sorry for the delayed response,

This works great!  I'm getting awesome speeds, but I've also tried to make both of the following optimization procedures work using the same code:
- SIMD using parallel data path inside a single CU.
- Multi-CU

I am finding that the same source code can do both individually:
- SIMD works when NUM_CU = 1
- Multi-CU works when SIMD ap_int<32> is used for floats.

When I turn both on at the same time, I get the following error in sw_emu

```bash
Found Platform
Platform Name: Xilinx
INFO: Reading /home/jenkins/repos/vitis_kernels/aten/src/ATen/fpga/torch.xclbin
Loading: '/home/jenkins/repos/vitis_kernels/aten/src/ATen/fpga/torch.xclbin'
8192
XRT build version: 2.3.1301
Build hash: 192e706aea53163a04c574f9b3fe9ed76b6ca471
Build date: 2019-10-24 20:05:16
Git branch: 2019.2
PID: 24004
UID: 1001
[Sat Mar 14 18:10:23 2020]
HOST: dylan-bespalko-ubuntu
EXE: /home/jenkins/repos/vitis_kernels/dev/BinaryOpsKernel/v_add/build/test
[XRT] WARNING: unaligned host pointer '0x55ba751f5840' detected, this leads to extra memcpy
[XRT] WARNING: unaligned host pointer '0x55ba751fd8c0' detected, this leads to extra memcpy
[XRT] WARNING: unaligned host pointer '0x55ba751e4520' detected, this leads to extra memcpy
corrupted double-linked list
Software emulation of compute unit(s) exited unexpectedly
Terminated
```

**Can you tell me if these features are mutually exclusive? They are somewhat redundant.**

closed 2020-03-15 by dylanbespalko

anjn, 2020-03-16:
Hi @dylanbespalko,

They are not mutually exclusive. Parallel operations in a kernel and multiple instantiations of that kernel make sense.