Xilinx/Vitis-AI #47: unrecognized Caffe layer type [Reshape]

GithubAbner, 2020-02-26:
When I use vitis-ai-caffe to compile my caffe model .
I got a err like:
"[VAI_C][Warning]Only 'SUM' type is supported for Eltwise,current layer is [stem_activation]"
"[VAI_C][Warning]Only 'SUM' type is supported for Eltwise,current layer is [block1a_activation]"
"[VAI_C][Error]unrecognized Caffe layer type [Reshape]"
"[VAI_C][Error]Parsing layer graph failed"
it looks like DPU does not support Reshape and Eltwise layer(except 'SUM'),but it difficults to change "Reshape" layer in my caffe model.
Is there some suggestions for me to avoid this error?
Thanks!

A1exy, 2020-03-31:
I also encountered the same error with the SSD model.
**************************************************
* VITIS_AI Compilation - Xilinx Inc.
**************************************************
[VAI_C][Warning] Only 'channel' axis supported on DPU for Concat, current layer is [mbox_priorbox].
[VAI_C][Error] Unrecognized Caffe layer type [Reshape].
[VAI_C][Error] Parsing layer graph failed.

qianglin-xlnx, 2020-05-11:
Hi GithubAbner,
I wonder if you have solved this issue. If not, I will submit this issue to the engineer in charge of it. If it has been resolved, please let me know and I will close this issue.
Note that if we do not receive any reply from you within 2 weeks, we will assume that this issue has been resolved and will close this issue. Thank you.

qianglin-xlnx, 2020-05-28:
Hi @GithubAbner and @A1exy ,
Yes, Reshape layer is not supported by the VAI_C, you have to remove Reshape layer from the prototxt and use CPU to process this layer. Hope it useful for you.

qianglin-xlnx, 2020-06-15:
Hi @GithubAbner 
Since I haven't received your reply for a long time, I will close this issue. If you still have a problem, you can reopen it. Thank you very much.

closed 2020-06-15 by qianglin-xlnx