Xilinx/Vitis-AI #50: Vitis ai tool docker  "sh 1_caffe_quantize.sh" operation failed

xumoxiao, 2020-03-06:
When i do "sh 1_caffe_quantize.sh",it's return that.

0306 01:41:38.437182   396 layer_factory.hpp:123] Creating layer data
I0306 01:41:38.437286   396 net.cpp:140] Creating Layer data
I0306 01:41:38.437325   396 net.cpp:455] data -> data
I0306 01:41:38.437364   396 net.cpp:455] data -> label
I0306 01:41:38.437518   396 image_data_layer.cpp:87] Opening file /home/xxx/fix_models_hrr_classification/val.txt
I0306 01:41:38.437568   396 image_data_layer.cpp:97] Shuffling data
I0306 01:41:38.437577   396 image_data_layer.cpp:102] A total of 0 images.
Segmentation fault (core dumped)

There are a few questions below.
1. Is the picture placed under path a xxxx/vitis-ai-tools-examples/images?
2. Must the name of the image begin with "ILSVRC2012_xxx"?

500 images in the images folder.
Where is the problem？

LinnHtunThaw, 2020-03-09:
I also encounter this problem, please advise.

qianglin-xlnx, 2020-05-11:
Hi @xumoxiao 
I wonder if you have solved this issue. If not, please describe your steps in detail and I will submit this issue to the engineer in charge of it. If it has been resolved, please close this issue.
Note that if we do not receive any reply from you within 2 weeks, we will assume that this issue has been resolved and will close this issue. Thank you.

qianglin-xlnx, 2020-06-21:
Hi @xumoxiao 
Since I haven't received your reply for a long time, I will close this issue. If you still have any questions, you can reopen this issue. Thank you very much.

closed 2020-06-21 by qianglin-xlnx