Xilinx/Vitis-AI #52: docker issue

savan1213, 2020-03-09:
To run this https://github.com/Xilinx/Vitis-AI/tree/master/alveo example  in the nimbix  i have give to github repo for dockerfile and json file.
So which repo i can use?

qianglin-xlnx, 2020-05-12:
Hi @savan1213 
I wonder if you have solved this issue. If not, please describe your issue in detail and I will submit this issue to the engineer in charge of it. If it has been resolved, please close this issue.
Note that if we do not receive any reply from you within 2 weeks, we will assume that this issue has been resolved and will close this issue. Thank you.

hanxue, 2020-05-12:
Could you point to which specific example that you are trying to run?

Please paste the prompt for github repo over here.

qianglin-xlnx, 2020-10-09:
hi @savan1213 
Since I haven't received your reply for a long time, I’m going to close this issue. Please feel free to reopen it if there are additional issues. Thank you very much.

closed 2020-10-09 by qianglin-xlnx