Xilinx/Vitis-AI #56: ERROR: [VPL 60-1328] Vpl run 'vpl' failed

sqPoseidon, 2020-03-16:
Hi, I failed to build dpu.xclbin of zcu102_base. I've already installed XRT and source the bash files before building the target. Here is the error message. Does anyone know how to solve this problem? Thank you in advance.

INFO: [v++ 60-1453] Command Line: vpl -t hw -f /home/qsun/app/Xilinx/Vitis/2019.2/platforms/zcu102_base/zcu102_base.xpfm --remote_ip_cache binary_container_1/ip_cache -s --output_dir /home/qsun/Compiler/Vitis-AI/DPU-TRD/prj/Vitis/binary_container_1/link/int --log_dir /home/qsun/Compiler/Vitis-AI/DPU-TRD/prj/Vitis/binary_container_1/logs/link --report_dir /home/qsun/Compiler/Vitis-AI/DPU-TRD/prj/Vitis/binary_container_1/reports/link --config /home/qsun/Compiler/Vitis-AI/DPU-TRD/prj/Vitis/binary_container_1/link/int/vplConfig.ini -k /home/qsun/Compiler/Vitis-AI/DPU-TRD/prj/Vitis/binary_container_1/link/int/kernel_info.dat --webtalk_flag Vitis --temp_dir /home/qsun/Compiler/Vitis-AI/DPU-TRD/prj/Vitis/binary_container_1/link --no-info --tlog_dir /home/qsun/Compiler/Vitis-AI/DPU-TRD/prj/Vitis/binary_container_1/.tlog/v++_link_dpu --iprepo /home/qsun/Compiler/Vitis-AI/DPU-TRD/prj/Vitis/binary_container_1/link/int/xo/ip_repo/xilinx_com_RTLKernel_sfm_xrt_top_1_0 --iprepo /home/qsun/Compiler/Vitis-AI/DPU-TRD/prj/Vitis/binary_container_1/link/int/xo/ip_repo/xilinx_com_RTLKernel_dpu_xrt_top_1_0 --messageDb /home/qsun/Compiler/Vitis-AI/DPU-TRD/prj/Vitis/binary_container_1/link/run_link/vpl.pb /home/qsun/Compiler/Vitis-AI/DPU-TRD/prj/Vitis/binary_container_1/link/int/dr.bd.tcl
INFO: [v++ 60-1454] Run Directory: /home/qsun/Compiler/Vitis-AI/DPU-TRD/prj/Vitis/binary_container_1/link/run_link

****** vpl v2019.2 (64-bit)
  **** SW Build 2708876 on Wed Nov  6 21:39:14 MST 2019
    ** Copyright 1986-2019 Xilinx, Inc. All Rights Reserved.

INFO: [VPL 60-839] Read in kernel information from file '/home/qsun/Compiler/Vitis-AI/DPU-TRD/prj/Vitis/binary_container_1/link/int/kernel_info.dat'.
INFO: [VPL 60-423]   Target device: zcu102_base
INFO: [VPL 60-1032] Extracting hardware platform to /home/qsun/Compiler/Vitis-AI/DPU-TRD/prj/Vitis/binary_container_1/link/vivado/vpl/.local/hw_platform
[23:32:28] Run vpl: Step create_project: Started
[23:32:29] Run vpl: Step create_project: Failed
[23:32:29] Run vpl: FINISHED. Run Status: create_project ERROR
ERROR: [VPL 60-1328] Vpl run 'vpl' failed
ERROR: [VPL 60-806] Failed to finish platform linker
INFO: [v++ 60-1442] [23:32:29] Run run_link: Step vpl: Failed
Time (s): cpu = 00:00:07 ; elapsed = 00:00:08 . Memory (MB): peak = 674.367 ; gain = 0.000 ; free physical = 4080 ; free virtual = 59475
ERROR: [v++ 60-661] v++ link run 'run_link' failed
ERROR: [v++ 60-626] Kernel link failed to complete
ERROR: [v++ 60-703] Failed to finish linking
Makefile:70: recipe for target 'binary_container_1/dpu.xclbin' failed
make: *** [binary_container_1/dpu.xclbin] Error 1

sqPoseidon, 2020-03-16:
More info in runme.log:

*** Running vivado
     with args -log vivado.log -applog -m64 -messageDb vivado.pb -mode batch -source vpl.tcl -notrace

 ****** Vivado v2018.2 (64-bit)
   **** SW Build 2258646 on Thu Jun 14 20:02:38 MDT 2018
   **** IP Build 2256618 on Thu Jun 14 22:10:49 MDT 2018
     ** Copyright 1986-2018 Xilinx, Inc. All Rights Reserved.
 
source vpl.tcl -notrace
[OPTRACE]|27983|1|/home/qsun/Compiler/Vitis-AI/DPU-TRD/prj/Vitis/binary_container_1/link/vivado/vpl/vpl.tcl|vivado_impl|1584372748985|START|Implementation|R    OLLUP_1
[23:32:28] Run vpl: Step create_project: Started
[23:32:29] Run vpl: Step create_project: Failed
couldn't load file "librdi_platformtcl.so": /home/qsun/app/Xilinx/Vitis/2019.2/lib/lnx64.o/librdi_platformtcl.so: undefined symbol: _ZTVN3sdx8platform19Nati    veObjectServiceE
INFO: [Common 17-206] Exiting Vivado at Mon Mar 16 23:32:29 2020...

qianglin-xlnx, 2020-05-11:
Hi @sqPoseidon 
I wonder if you have solved this issue. If not, please describe your steps in detail and I will submit this issue to the engineer in charge of it. If it has been resolved, please close this issue.
Note that if we do not receive any reply from you within 2 weeks, we will assume that this issue has been resolved and will close this issue. Thank you.

qianglin-xlnx, 2020-05-29:
Hi @sqPoseidon 
The following log show that you use Vivado v2018.2 to compile the zcu102_base. Do you have two versions of Vivado in your system?  Could you please check it again to make sure Vivado v2019.2 working. 
Let me know if you have any problem. Thank you. 

*** Running vivado
with args -log vivado.log -applog -m64 -messageDb vivado.pb -mode batch -source vpl.tcl -notrace

******** Vivado v2018.2 (64-bit)**
**** SW Build 2258646 on Thu Jun 14 20:02:38 MDT 2018
**** IP Build 2256618 on Thu Jun 14 22:10:49 MDT 2018
** Copyright 1986-2018 Xilinx, Inc. All Rights Reserved.

qianglin-xlnx, 2020-06-21:
Hi @sqPoseidon 
Since I haven't received your reply for a long time, I will close this issue. If you still have a problem, you can reopen it. Thank you very much.

closed 2020-06-21 by qianglin-xlnx