Xilinx/Vitis-AI #57: Limitations in input shapes

mpv89, 2020-03-17:
Hello -

My research group is trying to use Graph Neural Networks with Alveo cards and I noticed that the tools provided don't provide the flexibility we need.

I'm right now trying to quantitize a model that takes 4 different inputs with different shapes but the tensorflow tool (vai_q_tensorflow quantize) seems to only like to receive input shapes in 4 dimensions such as: ?,a,b,c. This is ideal for images but not for what we want to do.

I checked the source code and this seems to be hard-coded. Are there plans to expand the tools to include other input shapes?

As an example, this is the input shape I need for the GNNs we use: ?,3 : ?,4 : ? : ?

Thanks!

qianglin-xlnx, 2020-05-28:
Hi @mpv89 
 Yes, the tool has limitations of the inputs in vai1.0 and vai1.1, they must have 4 dimension shapes with format of ?,a,b,c.
Different dimension shapes will be supported in vai1.2.

closed 2020-05-28 by qianglin-xlnx

kavyab11, 2022-01-06:
Hi, I am trying to use the vai tensorflow quantiser to quantise a custom model consisting of fully connected layers. The model has multiple input nodes whose shapes are 2 dimensional (like ?,400 : ?,100 : ?,100). 

I have tried using both tensorflow and tensorflow2 quantisers.
- vai_q_tensorflow issue: Accepts only 4 dimensional inputs 
- vai_q_tensorflow2 issue: Accepts 2 dimensional input but only accepts a single input node

I have ensured that I am using the updated versions of all tools in Vitis AI. I am using the vai1.2 version from Vitis AI 1.4. I would like to know if there is any way to quantise a model which has multiple 2-dimensional input nodes which are passed on to dense layers consisting of 2D weight matrices, like in the example mentioned above.

Thanks!

AfifaIshtiaq, 2022-05-04:
Hi I'm working with Tensorflow2.x. I have calibration data in form of .npz format. Each .npz file has 6 inputs for the model. The input needs concatenated in a chunk of 2 [in1,in2] [in3,in4] [in5,in6] to be given to the actual model. I'm having difficulty in understading how should I call vitis_ai_quantizer with these multiple inputs ? what will be the command and steps that i need to quatize my model.