Xilinx/Vitis-AI #58: OpenCV in Conda is not support GUI like imshow().

lllkiller-jlll, 2020-03-20:
Hi.

I did alveo example "adas_detection" and I made sure that result images write on output directory.

But I can't see result images in code by cv::imshow().

The terminal print below:
terminate called after throwing an instance of 'cv::Exception'
  what():  OpenCV(3.4.2) /tmp/build/80754af9/opencv-suite_1535558553474/work/modules/highgui/src/window.cpp:632: error: (-2:Unspecified error) The function is not implemented. Rebuild the library with Windows, GTK+ 2.x or Carbon support. If you are on Ubuntu or Debian, install libgtk2.0-dev and pkg-config, then re-run cmake or configure script in function 'cvShowImage'

I think, the openCV seems like not support GUI environment in conda 'vitis-ai-caffe'.

How can I fix this error?

qianglin-xlnx, 2020-05-11:
Hi @lllkiller-jlll 
I wonder if you have solved this issue. If not, please describe your steps in detail and I will submit this issue to the engineer in charge of it. If it has been resolved, please close this issue.
Note that if we do not receive any reply from you within 2 weeks, we will assume that this issue has been resolved and will close this issue. Thank you.

lllkiller-jlll, 2020-05-12:
Hi @qianglin-xlnx 

I changed the source code in "main.cc" below.

`void displayFrame(bool &is_displaying) {
    Mat frame;
    static int fm_cnt=0;
    while (is_displaying) {
        mtxQueueShow.lock();
        if (queueShow.empty()) {
            if (is_running) {
                mtxQueueShow.unlock();
                usleep(10);
            } else {
                is_displaying = false;
                break;
            }
        } else if (idxShowImage == queueShow.top().first) {
            auto show_time = chrono::system_clock::now();
            stringstream buffer;
            frame = queueShow.top().second;
            auto dura = (duration_cast<microseconds>(show_time - start_time)).count();
            buffer << fixed << setprecision(1)
                   << (float)queueShow.top().first / (dura / 1000000.f);
            string a = buffer.str() + " FPS";
            cv::putText(frame, a, cv::Point(10, 15), 1, 1, cv::Scalar{240, 240, 240},1);
            
	    char img_name[20];
            sprintf(img_name,"output/image%d.jpg",fm_cnt); // writing inference output as image in output folder
            cv::imwrite(img_name,frame);
	    if(fm_cnt==0)
	    	cout<<"Saving results to output folder as .jpg images "<<endl;
            idxShowImage++;
            fm_cnt++;
            queueShow.pop();
            mtxQueueShow.unlock();
        } else {
            mtxQueueShow.unlock();
        }
    }
}`

to 

`void displayFrame(bool &is_displaying) {
    Mat frame;
    static int fm_cnt=0;
    while (is_displaying) {
        mtxQueueShow.lock();
        if (queueShow.empty()) {
            if (is_running) {
                mtxQueueShow.unlock();
                usleep(10);
            } else {
                is_displaying = false;
                break;
            }
        } else if (idxShowImage == queueShow.top().first) {
            auto show_time = chrono::system_clock::now();
            stringstream buffer;
            frame = queueShow.top().second;
            auto dura = (duration_cast<microseconds>(show_time - start_time)).count();
            buffer << fixed << setprecision(1)
                   << (float)queueShow.top().first / (dura / 1000000.f);
            string a = buffer.str() + " FPS";
            cv::putText(frame, a, cv::Point(10, 15), 1, 1, cv::Scalar{240, 240, 240},1);
            
	    char img_name[20];
            sprintf(img_name,"output/image%d.jpg",fm_cnt); // writing inference output as image in output folder
            
            //cv::imwrite(img_name,frame);
            cv::imshow("images", frame);
	    if(fm_cnt==0)
	    	cout<<"Saving results to output folder as .jpg images "<<endl;
            idxShowImage++;
            fm_cnt++;
            queueShow.pop();
            mtxQueueShow.unlock();
        } else {
            mtxQueueShow.unlock();
        }
    }
}`

I just changed imwrite to imshow.

And I received error message 'cv::Exception' I told.

Please submit this issue.

Thanks.

qianglin-xlnx, 2020-05-12:
Hi @lllkiller-jlll 
Please kindly provide some details. 
Which platform example are you running?  Alveo? alveo/examples/vitis_ai_alveo_samples/adas_detection?
Do you use Vitis AI 1.0 cpu docker or Vitis AI 1.0 gpu docker?

lllkiller-jlll, 2020-05-12:
Hi @qianglin-xlnx 

Yes. I did adas_example located in /alveo/examples/vitis_ai_alveo_samples/adas_detection.
And I used Vitis AI for CPU.

hanxue, 2020-08-03:
If you are trying to run GUI apps from within the Docker environment , you will need to modify the https://github.com/Xilinx/Vitis-AI/blob/master/docker_run.sh  file to have X Forwarding. In your GUI session, ensure the `$DISPLAY` environment variable is set, then add the following line

```
-e DISPLAY=$DISPLAY -v /tmp/.X11-unix:/tmp/.X11-unix:rw
```

As an example, in the `docker_run.sh` , run the following 

```
  docker run \
    -e DISPLAY=$DISPLAY -v /tmp/.X11-unix:/tmp/.X11-unix:rw
    $docker_devices \
    -v /opt/xilinx/dsa:/opt/xilinx/dsa \
    -v /opt/xilinx/overlaybins:/opt/xilinx/overlaybins \
    -e USER=$user -e UID=$uid -e GID=$gid \
    -v /dev/shm:/dev/shm \
    -v $HERE:/workspace \
    -w /workspace \
    -it \
    --rm \
    --network=host \
    $IMAGE_NAME \
    bash
```

Some further reference for running X applications from within Docker container

https://stackoverflow.com/a/25334301/399759
http://wiki.ros.org/docker/Tutorials/GUI

Have you been able to get GUI output from the Vitis AI CPU docker container?

lllkiller-jlll, 2020-10-14:
Thanks for your reply @hanxue .

I solved this problem.

I'll close this issue.

closed 2020-10-14 by lllkiller-jlll