Xilinx/Vitis-AI #59: Illegal variable name

jing-fi, 2020-03-20:
Hi 
I am running Vitis-AI/alveo/examples/tensorflow/ examples
(https://github.com/Xilinx/Vitis-AI/tree/master/alveo/examples/tensorflow)

however in the step source setup.sh
source /home/jingy/Vitis-AI/alveo/overlaybins/setup.sh
shows error "Illegal variable name"

I am using ubuntu16.04, and Alveo U200 board.
How do i fix the issue?

sumitn-xilinx, 2020-03-20:
Did you load the docker and activated the corresponding conda environement?

jing-fi, 2020-03-20:
> CAUTION: This message has originated from an External Source. Please use proper judgment and caution when opening attachments, clicking links, or responding to this email.

Hi ,

Do you mean this one?
https://github.com/Xilinx/Vitis-AI/blob/master/doc/install_docker/load_run_docker.md
cpu tools docker, gpu tools docker and runtime docker?

If yes, I haven’t load it since I am not sure the version.
May I know how do I check the cpu, gpu and runtime version?

Regards,
Yang Jing

sumitn-xilinx, 2020-03-20:
try xilinx/vitis-ai:runtime-1.0.0-cpu

jing-fi, 2020-03-20:
It is correct?

jingy@xapthor40:/home/jingy/Vitis-AI% ./docker_run.sh xilinx/vitis-ai:tools-1.0.0-cpu
Running as jingy
passwd: password expiry information changed.
To run a command as administrator (user "root"), use "sudo <command>".
See "man sudo_root" for details.

bash: export: `1': not a valid identifier
bash: export: `14:51:32': not a valid identifier
bash: export: `2019': not a valid identifier

==========================================
__      ___ _   _                   _____
\ \    / (_) | (_)            /\   |_   _|
\ \  / / _| |_ _ ___ ______ /  \    | |
  \ \/ / | | __| / __|______/ /\ \   | |
   \  /  | | |_| \__ \     / ____ \ _| |_
    \/   |_|\__|_|___/    /_/    \_\_____|

==========================================

Docker Image Version: 1.0.0
Build Date: Sun
VAI_ROOT=/opt/vitis_ai
For TensorFlow Workflows do:
  conda activate vitis-ai-tensorflow
For Caffe Workflows do:
  conda activate vitis-ai-caffe
jingy@xapthor40:/workspace$

jing-fi, 2020-03-20:
Not sure how to fix this

jingy@xapthor40:/home/jingy/Vitis-AI%                                                         D
jingy@xapthor40:/home/jingy/Vitis-AI% sudo ./docker_run.sh xilinx/vitis-ai:runtime-1.0.0-cpu
[sudo] password for jingy:
WARNING: Error loading config file: /home/jingy/.docker/config.json: stat /home/jingy/.docker/config.json: permission denied
Running as root
useradd: user 'root' already exists
/etc/login.sh: line 12: /home/root/.bashrc: No such file or directory
/etc/login.sh: line 13: /home/root/.bashrc: No such file or directory

hanxue, 2020-08-03:
Hi @jing-fi , could you please try with the latest v1.2 docker images?

The summarized steps are:
1. Download and install Docker (version 19.03.1 preferably)
2. Create the `docker` group and add yourself to it
3. Get a copy of this repo 

```
git clone --recurse-submodules https://github.com/Xilinx/Vitis-AI  
cd Vitis-AI
```

Or if you already have a copy, then update it

```
cd Vitis-AI
git pull origin master
```

4. Get the latest Vitis AI 1.2 CPU docker image `docker pull xilinx/vitis-ai:latest `
5. Run the docker image

```
./docker_run.sh xilinx/vitis-ai:latest
```

What is the result of using the v1.2 docker image?

qianglin-xlnx, 2020-12-11:
Hi @jing-fi 
Since we haven't received your reply for a long time, I'm going to close this issue. If you still have any questions, please feel free to reopen it. Thank you very much.

closed 2020-12-11 by qianglin-xlnx