Xilinx/Vitis-AI #64: xclbin not provided  for  "whole_app_acceleration"

mahinlma, 2020-03-24:
@wilderfield @sumitn-xilinx @kamranjk 
Regarding this 
https://github.com/Xilinx/Vitis-AI/tree/master/alveo/apps/whole_app_acceleration

Already discussed the same(xplusml) issue here
https://github.com/Xilinx/Vitis-AI/issues/41

And also please provide xclbin for  latest XRT version 2019.2

mahinlma, 2020-03-24:
@sumitn-xilinx @wilderfield Please let me know, How to utilize libs provided 
https://github.com/Xilinx/Vitis-AI/tree/master/alveo/apps/whole_app_acceleration/libs

sumitn-xilinx, 2020-03-25:
you first need the new xclbins. we will update this thread soon.

mahinlma, 2020-03-25:
Okay, Thank you. @sumitn-xilinx , Please provide  xclbin for XRT version 2019.2

nkpavan, 2020-03-25:
@mahinlma , you can download the xclbins using below links.
for Ubuntu:
https://www.xilinx.com/bin/public/openDownload?filename=vai-1.1-xplusml-bins-18.04.deb
for centOS:
https://www.xilinx.com/bin/public/openDownload?filename=vai-1.1-xplusml-bins-1.0-0.x86_64.rpm

mahinlma, 2020-03-25:
Thank you @nkpavan

closed 2020-03-25 by mahinlma