Xilinx/Vitis-AI #65: Error while running - AI Kernel Scheduler

mahinlma, 2020-03-24:
@wilderfield  @sumitn-xilinx 

Please let me know, how to fix this issue.
[ERROR] libdpuv1utils.so: cannot open shared object file: No such file or directory
[ERROR] Could not find the mandatory 'getKernel' function in libs/libYoloPostProc.so. Skipping this kernel.

sumitn-xilinx, 2020-03-25:
which docker are you using?

mahinlma, 2020-03-25:
@sumitn-xilinx ,
docker from this https://hub.docker.com/r/xilinx/vitis-ai/tags

Please share the recent docker link

mahinlma, 2020-03-25:
@sumitn-xilinx , Referred here https://github.com/Xilinx/Vitis-AI/blob/master/doc/install_docker/load_run_docker.md

sumitn-xilinx, 2020-03-25:
please wait for the new docker to use AI Kernel Scheduler. Should be available in a day or so.

mahinlma, 2020-03-25:
Thank you @sumitn-xilinx

closed 2020-03-25 by mahinlma