Xilinx/Vitis-AI #66: Download of vitis_ai_runtime_library_r1.1 does not have .deb files for runtime package.

liweintu, 2020-03-25:
Hi there, I'm following the latest Vitis-AI/VART/README.md (from commit: 4ea2380) for setting up ZCU102 in Vitis AI, such that I can cross-compile my network models on host and run them on ZCU102.
However, when I tried to scp the .deb files to the ZCU102 board, I found the downloaded tarball didn't have them. Only two folders were there in the tarball, samples and demo, which contained many .jpg images.

Is the download link wrong?

Please note that I've previously set up the SD card using petalinux and examples from folder Vitis-AI/mpsoc/vitis-ai-tool-example/ have successfully run on ZCU102.

busche, 2020-04-03:
I finally succeeded in getting the ZCU102 up and running.

I was downloading
wget https://www.xilinx.com/bin/public/openDownload?filename=vitis-ai-runtime-1.1.0.tar.gz

Therein, the .deb packages are somewhat "hidden" in inside subdirs. You can easily find them, though.

I did also notice a link mismatch between the official documentation at
https://www.xilinx.com/html_docs/vitis_ai/1_1/yjf1570690235238.html
and the information here on github. Following the github documentation helped me to succeed.

Best luck!

SusmithaGadagottu, 2020-04-17:
Hi Busche,

I have a zcu102 evaluation board and I have setup the board by following the instructions from quick start and also readme file on github. I am now trying to run samples from:

Now I see below errors when I run the sample applications. Did you see any issues like this or are you aware of this error?
Error when running resnet50 from VART folder:

WARNING: Logging before InitGoogleLogging() is written to STDERR
F0416 17:53:44.919674 3203 xrt_cu.cpp:156] Check failed: is_done cu timeout! core_idx 1 handle=0x557695b1c0 ENV_PARAM(XLNX_DPU_TIMEOUT) 10000 state 1 ERT_CMD_STATE_COMPLETED 4 ms 10010 bo=1is_done 0
*** Check failure stack trace: ***
Aborted

Same issue happens when the run the sample applications from the overview folder.

~/overview/samples/classification# ./test_jpeg_classification resnet50 sample_classification.jpg

WARNING: Logging before InitGoogleLogging() is written to STDERR
F0416 17:19:27.592169 3193 xrt_cu.cpp:156] Check failed: is_done cu timeout! core_idx 1 handle=0x559b2d1800 ENV_PARAM(XLNX_DPU_TIMEOUT) 10000 state 1 ERT_CMD_STATE_COMPLETED 4 ms 10010 bo=1is_done 0
*** Check failure stack trace: ***
Aborted

Thanks
Susmitha

qianglin-xlnx, 2020-05-11:
Hi @liweintu,
I wonder if you have solved this issue. If yes, could you please close this issue? If not, please also let me know and I will confirm this issue.
Note that if we do not receive any reply from you within 2 weeks, we will assume that this issue has been resolved and will close this issue. Thank you.

liweintu, 2020-05-11:
Thanks guys for the suggestions, I actually haven't got time to verify the solutions yet due to being switched to some other urgent tasks. Anyway, I can close it for now and will follow up with this fix later. If the issue still exists, we can consider re-opening this later.

closed 2020-05-11 by liweintu