Xilinx/Vitis-AI #68: Issue - No response with Googlenet-v1 without jpeg decoder

mahinlma, 2020-03-25:
@nkpavan @sumitn-xilinx @wilderfield 

Worked fine with resnet50 with JPEG decoder
> Running Inference with HW Pre-processing with JPEG decoder
> -------------------
> Speaking to Butler 
> Response from Butler is: 
> errCode: errCode: 0
> errCode String: SUCCESS
> myHandle: 1
> valid: 1
> 
> [XDNN] loading xclbin settings from /opt/xilinx/overlaybins/xdnnv3/xplusml/classification/with_jpeg_decoder/xdnn_v3_96x16_2pe_8b_9mb_bank03.xclbin.json
> [XDNN] using custom DDR banks 0,3
> Path /tmp/vitis-ifpga/76/weights.h5 is a file.
> Loading weights/bias/quant_params to FPGA...
> 
> [XRT]    git hash                   : 192e706aea53163a04c574f9b3fe9ed76b6ca471
> [XDNN]   git hash                   : 6665f90940d49ab8337893307a87aa1afd4db13e
> [XDNN] kernel configuration
> [XDNN]   num cores                  : 2
> [XDNN]   dsp array width            : 96
> [XDNN]   axi data width (in 32bits) : 16
> [XDNN]   img mem size               : 9 MB
> [XDNN]   max instr num              : 1536
> [XDNN]   max xbar entries           : 4096
> [XDNN]   version                    : 3.2
> [XDNN]   8-bit mode                 : 1
> ---------- Prediction 1/1 for /workspace/alveo/apps/whole_app_acceleration/classification/dog.jpg ----------
> 0.9433 "n02099601 golden retriever"
> 0.0217 "n02106030 collie"
> 0.0060 "n02088094 Afghan hound, Afghan"
> 0.0048 "n02091831 Saluki, gazelle hound"
> 0.0043 "n02090622 borzoi, Russian wolfhound"
> 
> 
> 

**Issue:**

When executed for googlenet-v1, no response
`./run.sh `

> Response from Butler is: 
> errCode: errCode: 0
> errCode String: SUCCESS
> myHandle: 2
> valid: 1
> 
> Queue populated
> [XDNN] loading xclbin settings from /opt/xilinx/overlaybins/xdnnv3/xplusml/classification/wo_jpeg_decoder/xdnn_v3_96x16_2pe_8b_9mb_bank03.xclbin.json
> [XDNN] using custom DDR banks 0,3
> Path /tmp/vitis-ifpga/195/weights.h5 is a file.
> Loading weights/bias/quant_params to FPGA...
> 
> [XRT]    git hash                   : 192e706aea53163a04c574f9b3fe9ed76b6ca471
> [XDNN]   git hash                   : 6665f90940d49ab8337893307a87aa1afd4db13e
> [XDNN] kernel configuration
> [XDNN]   num cores                  : 2
> [XDNN]   dsp array width            : 96
> [XDNN]   axi data width (in 32bits) : 16
> [XDNN]   img mem size               : 9 MB
> [XDNN]   max instr num              : 1536
> [XDNN]   max xbar entries           : 4096
> [XDNN]   version                    : 3.2
> [XDNN]   8-bit mode                 : 1

durgabhavaniv, 2020-04-21:
@mahinlma @nkpavan @sumitn-xilinx @wilderfield

Able to get response for classification with Googlenet-v1 model without jpeg decoder without any modification in the original code.

Results are as follows.

with Googlenet-v1 model

[image]

with resnet50 model

[image]

regards,
V Durga.

durgabhavaniv, 2020-04-21:
@mahinlma can close the issue.

mahinlma, 2020-04-21:
Forgot to close this issue. Received response on the same day.

closed 2020-04-21 by mahinlma