Xilinx/Vitis-AI #71: bash: ./demo_classification: cannot execute binary file: Exec format error

hopesala, 2020-03-26:
I run the demo in the Vitis-AI-Library/overview/demo/yolo3 and classification, 
first sh build.sh, then ./demo_yolov3 xxx.jpg, 
but got the same error:
bash: ./demo_yolov3: cannot execute binary file: Exec format error
bash: ./demo_classification: cannot execute binary file: Exec format error

what is the problem with this, anyone can help? thanks ahead

~ /Desktop/Vitis-AI/Vitis-AI-Library/overview/demo/classification$ sh build.sh 
caoche@ubuntu:~ /Desktop/Vitis-AI/Vitis-AI-Library/overview/demo/classification$ ls
build.sh  demo_classification  demo_classification.cpp  readme
caoche@ubuntu:~ /Desktop/Vitis-AI/Vitis-AI-Library/overview/demo/classification$ ./demo_classification ../yolov3/people_and_dog.jpg 
bash: ./demo_classification: cannot execute binary file: Exec format error

qianglin-xlnx, 2020-05-11:
Hi @hopesala,
For edge, when you cross compile the program on the host, you have to run it on the target board, such as ZCU102 or ZCU104. You can't run it on your host side directly.

qianglin-xlnx, 2020-05-28:
Hi @hopesala 
May I know if your issue has been solved? Let me know if you have any problems. Thank you.

qianglin-xlnx, 2020-06-15:
Hi @hopesala 
Since I haven't received your reply for a long time, I will close this issue. If you still have a problem, you can reopen it. Thank you very much.

closed 2020-06-15 by qianglin-xlnx

tyler274, 2023-07-05:
Still getting this even though i followed the instructions for cross compiling and am running it on the board.

quentonh, 2023-08-10:
As a first step for users who encounter this issue, I would recommend double-checking your CXX variable to confirm that it is correctly set for the appropriate architecture.

quentonh@XSJ-QUENTONH-L1:/mnt/c/Users/quentonh$ echo $CXX

aarch64-xilinx-linux-g++ -mcpu=cortex-a72.cortex-a53 -march=armv8-a+crc -fstack-protector-strong -O2 -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security --sysroot=/home/quentonh/petalinux_sdk_2022.2/sysroots/cortexa72-cortexa53-xilinx-linux