Xilinx/Vitis-AI #84: Choose what to install with the `sdk.sh`

finn-ball, 2020-04-15:
At the moment it installs binaries for both `aarch64` and `x86_64`. It may be useful to allow the user to choose to only install what they need.

qianglin-xlnx, 2020-05-13:
Hi @finn-ball 
It's related to sysroot which is shipped in SDK form. It is a standard form that yocto provides.

closed 2020-05-28 by qianglin-xlnx

finn-ball, 2020-05-28:
Did this get closed because it was resolved?

qianglin-xlnx, 2020-05-28:
It's a standard form that yocto provides. Both the files under aarch64 and x86_64 are used for crossing compiling. So It's not possible to choose what to install with the 'sdk.sh'. 
Thank you for your advice again.

finn-ball, 2020-05-28:
That's unusual. When I cross compile, I generally just need a cross compiler, along with the libraries in the `aarch64` folder. Is this not the case for Xilinx?

qianglin-xlnx, 2020-05-28:
The cross compiler tools, such as aarch64-xilinx-linux-g++ and aarch64-xilinx-linux-gcc, are under 
./x86_64-petalinux-linux/usr/bin/aarch64-xilinx-linux/
They can be run on X86 host.  They are not under aarch64 which makes you confused I guess.
[image]