Xilinx/Vitis-AI #85: Issues running sample examples on zcu102 with Vitis AI

SusmithaGadagottu, 2020-04-17:
Hi,

I have a zcu102 evaluation board and I am trying to run the samples from Running Vitis AI Examples Based on VART.
 I have setup the host and evaluation board following the instructions from below locations.
https://www.xilinx.com/html_docs/vitis_ai/1_1/lbv1583226048322.html
https://www.xilinx.com/html_docs/vitis_ai/1_1/aqc1570690254969.html

Now I see below errors when I run the sample application. Could you please help me resolve the error?

./resnet50 model_dir_for_zcu102/

WARNING: Logging before InitGoogleLogging() is written to STDERR
F0416 17:53:44.919674  3203 xrt_cu.cpp:156] Check failed: is_done cu timeout! core_idx 1  handle=0x557695b1c0 ENV_PARAM(XLNX_DPU_TIMEOUT) 10000 state 1 ERT_CMD_STATE_COMPLETED 4 ms 10010  bo=1is_done 0 
*** Check failure stack trace: ***
Aborted

Same issue happens when the run the sample applications from the overview folder.

~/overview/samples/classification# ./test_jpeg_classification resnet50 sample_classification.jpg 

WARNING: Logging before InitGoogleLogging() is written to STDERR
F0416 17:19:27.592169  3193 xrt_cu.cpp:156] Check failed: is_done cu timeout! core_idx 1  handle=0x559b2d1800 ENV_PARAM(XLNX_DPU_TIMEOUT) 10000 state 1 ERT_CMD_STATE_COMPLETED 4 ms 10010  bo=1is_done 0 
*** Check failure stack trace: ***
Aborted

Thanks
Susmitha

qianglin-xlnx, 2020-05-11:
Hi @SusmithaGadagottu

Could you follow the instructions on the links below and try it again?
https://github.com/Xilinx/Vitis-AI/tree/master/VART#setting-up-the-host-for-edge 
https://github.com/Xilinx/Vitis-AI/tree/master/VART#setting-up-the-target
https://github.com/Xilinx/Vitis-AI/tree/master/VART#running-vitis-ai-examples-for-edge
#Flashing the OS Image to the SD Card
https://www.xilinx.com/html_docs/vitis_ai/1_1/gum1570690244788.html

qianglin-xlnx, 2020-05-28:
Hi @SusmithaGadagottu 
May I know if your issue has been solved? Let me know if you have any problems. Thank you.

SusmithaGadagottu, 2020-05-28:
I am going to try this tomorrow and let you know the status by end of day
tomorrow.

lhuipeng, 2020-06-11:
Hi @SusmithaGadagottu and @qianglin-xlnx 

I also encountered the same error when following the instructions on the links 

https://github.com/Xilinx/Vitis-AI/tree/master/VART#setting-up-the-host-for-edge
https://github.com/Xilinx/Vitis-AI/tree/master/VART#setting-up-the-target
https://github.com/Xilinx/Vitis-AI/tree/master/VART#running-vitis-ai-examples-for-edge
#Flashing the OS Image to the SD Card
https://www.xilinx.com/html_docs/vitis_ai/1_1/gum1570690244788.html

Error message is  
root@xilinx-zcu102-2019_2:~/VART/samples/resnet50# ./resnet50 model_dir_for_zcu102
WARNING: Logging before InitGoogleLogging() is written to STDERR
F1126 06:35:55.888309  3111 xrt_cu.cpp:156] Check failed: is_done cu timeout! core_idx 1  handle=0x556ca39f80 ENV_PARAM(XLNX_DPU_TIMEOUT) 10000 state 1 ERT_CMD_STATE_COMPLETED 4 ms 10010  bo=1is_done 0 
*** Check failure stack trace: ***
Aborted

Do you have idea how to resolve this ? 

regards,
HuiPeng

qianglin-xlnx, 2020-06-11:
Hi @lhuipeng 
It seems that the DPU works abnormal. It may caused by the OS image flashing abnormal or the unstable SD . I think you can try the following:
1. Use another SD, >=16G is better
2. Using Etcher tool to flash the OS image again .
https://www.xilinx.com/html_docs/vitis_ai/1_1/gum1570690244788.html

lhuipeng, 2020-06-16:
Hi @qianglin-xlnx,

With the new OS image, it is working now. Thanks.

regards,

Hui Peng

https://www.xilinx.com/html_docs/vitis_ai/1_1/gum1570690244788.html

SusmithaGadagottu, 2020-06-16:
I am ok now and able to run the samples from VART and also AI libraries.
This issue can be closed if it is not closed already.

closed 2020-06-16 by qianglin-xlnx