Xilinx/Vitis-AI #87: Need Guidance for Alveo U50

jaydevshelat, 2020-04-21:
Hello,

I am new to Xilinx Alveo and Git.  I have an Alveo U50 PCIe card.  I am able to install and bring-up the board successfully. 

I am looking for guidance on what is the best examples/links/references/apps for the Alveo U50 board. 

Looking around; it seems like most of the material is either applicable to U200/U250 cards or the ZCU boards. 

Thank You in advance!

mahinlma, 2020-04-23:
@jaydevshelat 
for U50 refer https://github.com/Xilinx/Vitis-AI/tree/master/Vitis-AI-Library#setting-up-the-host-for-alveo

jaydevshelat, 2020-04-24:
Hi @mahinlma 

Thank You for the link. I am able to perform that step. Also, I am able to run some of the examples/demo/samples available in Vitis-AI-Library/Overview/  as well as  VART/Samples 

Next Questions :  What is the quickest path to achieve following two objectives:

1. Using the Xilinx provided Models perform a Classification benchmark exercise (CPU-only vs. CPU+Alveo) on set of images
2. Use modified pre-trained model and repeat step 1

What is the Vitis-AI-Library/Benchmark for?  Will it help for step 1? IF yes, how?

Please let me know.  I appreciate your time & guidance!

mahinlma, 2020-04-25:
@jaydevshelat 
Glad to know that you are able to execute on U50.

@sumitn-xilinx , @wilderfield please give guidance to him.

sumitn-xilinx, 2020-04-27:
"Using the Xilinx provided Models perform a Classification benchmark exercise (CPU-only vs. CPU+Alveo) on set of images"
--> You can run the model on Caffe CPU. For Alveo, you can use following command in alveo/examples/deployment_modes for at least 50k images.
./run.sh \
  -t streaming_classify   \
  -d <Path to image dataset> \
  -cw $VAI_ALVEO_ROOT/examples/caffe/$WORK/deploy.caffemodel_data.h5 \
  -cn $VAI_ALVEO_ROOT/examples/caffe/$WORK/compiler.json \
  -cq $VAI_ALVEO_ROOT/examples/caffe/$WORK/quantizer.json \
 -y 16 \
 --profile

"Use modified pre-trained model and repeat step 1"
---> Same as above
"What is the Vitis-AI-Library/Benchmark for? Will it help for step 1? IF yes, how?"
--> Benchmark numbers mentioned for alveo U200/250 in AI Model Zoo (https://github.com/Xilinx/AI-Model-Zoo/) are captured using above method.

jaydevshelat, 2020-04-27:
@sumitn-xilinx :  Thank You!   One clarification:  Will this work on the U50 board?  

From my previous efforts;  the alveo/ folder seems to be only for U200 and U250 boards. 

Appreciate your help!

sumitn-xilinx, 2020-04-28:
Sorry, I missed the title. Alveo folder is only for U200/250. 
You will have to use Vitis-AI-Library for reproducing the numbers for U50. 
See section - Running Vitis AI Library Examples (For Cloud)

qianglin-xlnx, 2020-05-13:
Hi @jaydevshelat 
I don’t know whether your issue has been solved and whether you have successfully run the performance sample on U50 board. Please let us know if you are still encountering some issue.

closed 2021-07-22 by qianglin-xlnx