Xilinx/Vitis-AI #89: I want to use protobuf 3.5.0, but error happens when I integrate vitis ai into my project

etdjj, 2020-04-27:
Becasue of some reasones, I need to use protobuf 3.5.0 in my project. Then I add preferred version in petalinuxbsp.conf"
```
PREFERRED_VERSION_protobuf = "3.5.0"
PREFERRED_VERSION_protobuf-native = "3.5.0"
```
But some error happened during "do rootfs":
```
Problem: conflicting requests
nothing provides libprotobuf17 needed by packagegroup-petalinux-xlnx-ai-1.0-r0.noarch
```
[image]

qianglin-xlnx, 2020-05-12:
Hi @etdjj 
I wonder if you have solved this issue. If not, please describe your steps in detail and I will submit this issue to the engineer in charge of it. If it has been resolved, please close this issue.
Note that if we do not receive any reply from you within 2 weeks, we will assume that this issue has been resolved and will close this issue. Thank you.

qianglin-xlnx, 2020-06-21:
Hi  @etdjj 
The protobuf version required in the released Petalinux BSP is 3.6.1. Therefore, protobuf below 3.6.1 is not supported.

closed 2020-06-21 by qianglin-xlnx