Xilinx/Vitis-AI #93: Not able to run Tool-Example/4_tf_quantize.sh

durgabhavaniv, 2020-05-03:
Getting the following error:

INFO: Checking Float Graph...
Traceback (most recent call last):
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/bin/vai_q_tensorflow", line 11, in <module>
    sys.exit(run_main())
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/tensorflow_core/contrib/decent_q/python/decent_q.py", line 781, in run_main
    app.run(main=my_main, argv=[sys.argv[0]] + unparsed)
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/tensorflow_core/python/platform/app.py", line 40, in run
    _run(main=main, argv=argv, flags_parser=_parse_flags_tolerate_undef)
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/absl/app.py", line 299, in run
    _run_main(main, args)
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/absl/app.py", line 250, in _run_main
    sys.exit(main(argv))
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/tensorflow_core/contrib/decent_q/python/decent_q.py", line 780, in <lambda>
    my_main = lambda unused_args: main(unused_args, FLAGS)
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/tensorflow_core/contrib/decent_q/python/decent_q.py", line 576, in main
    flags.skip_check)
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/tensorflow_core/contrib/decent_q/python/decent_q.py", line 323, in quantize_frozen
    check_float_graph(input_graph_def, input_fn, q_config, s_config)
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/tensorflow_core/contrib/decent_q/python/decent_q.py", line 242, in check_float_graph
    inputs = input_fn(iter=0)
  File "./example_file/input_fn.py", line 83, in calib_input
    image = resize_shortest_edge(image, 256)
  File "./example_file/input_fn.py", line 11, in resize_shortest_edge
    H, W = image.shape[:2]
AttributeError: 'NoneType' object has no attribute 'shape'

foolmarks, 2020-05-03:
AttributeError: 'NoneType' object has no attribute 'shape'   is the classic error message that you get from OpenCV when you try to manipulate an image but the preceeding cv2.imread() command has failed (..unfortunately it fails silently..).

I suspect that the input_fn.py script is looking for images and not finding them - you may need to supply them from the ImageNet validation set - try downloading using Academic Torrents:

https://academictorrents.com/details/5d6d0df7ed81efd49ca99ea4737e0ae5e3a5f2e5

durgabhavaniv, 2020-05-03:
@foolmarks 

Thanks for response it worked

closed 2020-05-03 by durgabhavaniv