Xilinx/Vitis-AI #94: Why accuracy increased after quantization ?

durgabhavaniv, 2020-05-04:
In [https://github.com/Xilinx/Vitis-AI/tree/master/Tool-Example#vitis-ai-tool-example] the following results given.

FP32 evaluation : [https://github.com/Xilinx/Vitis-AI/tree/master/Tool-Example#float-model-evaluation]

INT8 evaluation : [https://github.com/Xilinx/Vitis-AI/tree/master/Tool-Example#quantized-model-evaluation]

durgabhavaniv, 2020-05-13:
I didn't get reply till now....

sumitn-xilinx, 2020-05-14:
@qianglin-xlnx  - Can you please repond to this query?

qianglin-xlnx, 2020-05-14:
@sumitn-xilinx Sure. In fact, I have submitted this issue to the person in charge of it and still waiting for a clear answer.
@durgabhavaniv From my point view, float point model is usually not optimally trained in this case. You can retrain your float point model for better accuracy , at lease 0.79 in this example.  Tool-Example just show users how to use the tools in Vitis AI.

qianglin-xlnx, 2020-05-14:
@durgabhavaniv 
The accuracy increase slightly because of the calib data is only a small portion of whole imagenet dataset. The accuracy varies a little after calibration and test. 

The  quantized model doesn't change in size because we didn't modify the data directly but only save quantize info in it, which will be used by compiler then.

qianglin-xlnx, 2020-05-28:
hi @durgabhavaniv 
Since I haven't received your reply for a long time, I will close this issue. If you still have any questions, you can reopen this issue. Thank you.

closed 2020-05-28 by qianglin-xlnx