Xilinx/Vitis-AI #95: Profiling and fine-grained profiling - Vitis AI 1.1

kerocc, 2020-05-04:
Hi all,

I have a few questions regarding the use of the profiler with Vitis 1.1. I work on the ZCU102 board. I followed the GitHub tutorial (https://github.com/Xilinx/Vitis-AI/blob/master/VART/README.md).

I do not manage to generate the dpu_trace_[PID].prof file on the resnet50 exemple (provided here: https://github.com/Xilinx/Vitis-AI/tree/master/VART/samples/resnet50).

- **1st question:** dexplorer seems not to be deployed on the image provided for the ZCU102. I copied/pasted the dexplorer executable from Vitis AI 1.0 runtime docker (which has been deprecated in Vitis AI 1.1 if I'm right). I copied dsight as well. Is it the right way to proceed, or should I find these executables somewhere else?

Then I run `dexplorer -m profile`.

Then I run the resnet50 `./resnet50 model_dir_for_zcu102`.
It runs well, but I have no .prof generated (I had the file when working with Vitis AI 1.0).

I manage to have the .prof files only with DNNDK examples (https://github.com/Xilinx/Vitis-AI/tree/master/mpsoc/vitis_ai_dnndk_samples/resnet50), after following this guide : https://github.com/Xilinx/Vitis-AI/blob/master/mpsoc/README.md. These are the only examples using the C++ APIs, so I guess this is why the profiler works here? Aren't these supposed to be deprecated over time since I understood Vitis AI "replaces" DNNDK ?

- **2nd question** then: is it possible to use the profiler based on the VART samples syntax/APIs, or is it mandatory to use the dnndk lib (dnndk/dnndk.h) in order to enable the profiler?

Last but not least is about the fine-grained profiler.
From UG1414 (v1.0) December 18, 2019 (Vitis AI User Guide), page 93, I read that I have to use a dedicated flag (T_MODE_PROF) when calling the dpuCreateTask() function in order to enable the fine-grained profiling.

In the provided file is the line 230:
`taskResnet50 = dpuCreateTask(kernelResnet50, 0);`
0 stands for T_MODE_NORMAL (according to https://www.xilinx.com/html_docs/vitis_ai/1_1/dao1576168491848.html). I tried to replace it by T_MODE_PROF but it does not compile then (variable not declared). I tried to replace it by 1 (supposing it matches T_MODE_PROF?) but the resulting .prof file is identical to the one generated with default "0" T_MODE_NORMAL flag.

- So my **3rd question** is: how do I enable fine-grained profiling? Since the executable outputs the same thing as UG1414 Figure 34 page 94, is the fine-grained profiling enabled by default today?

I would like to be able to generate a nice .html file, showing the impact of each layer of the neural network, not only the DPU core utilization over time.
But is it available yet? I just found this topic while writing my question: https://forums.xilinx.com/t5/AI-and-Vitis-AI/Vitis-AI-profiler/td-p/1097135 which would end pretty much all of my questions about the fine-grained profiler...

Thanks for your help and the time you dedicated to this post, do no hesitate to ask me if I have not been clear enough in my explanations :-)

Regards

foolmarks, 2020-05-05:
Dexplorer and profiling are not currently available with the new VART runtime. They will be introduced in a future release.

qianglin-xlnx, 2020-07-24:
Hi @kerocc 
Vai1.2 has been released and Vitis-AI-Profiler is available now. You can use it to profile the sapmles based on VART runtime.

hanxue, 2020-08-03:
Hi @foolmarks , are you able to generate fine-grained profiling now with Vitis-AI-Profiler on v1.2?

kerocc, 2020-08-10:
> 
> 
> Hi @kerocc
> Vai1.2 has been released and Vitis-AI-Profiler is available now. You can use it to profile the sapmles based on VART runtime.

Hi @qianglin-xlnx 
Thanks for the release. 

I did the following and managed to use the new detailed profiler (maybe this will answer your question @hanxue):

1. I adjusted my personal Vitis AI 1.1 VART application to match the new APIs (basically I looked at the differences between 1.1 and 1.2 VART samples and did the same changes in my code).

2. I just ran the following command on the board (ZCU102 here):
`/usr/bin/xlnx/vaitrace/vaitrace.py -t 10 -o ~/my_trace.xat ./my_VART_app <my_VART_app_args>`

3. The previous command creates the _my_trace.xat_ as expected (and later I visualize the .xat in browser, following these instructions: https://github.com/Xilinx/Vitis-AI/tree/master/Vitis-AI-Profiler).

I still need to analyse the results but at least I have the full profiling flow working. One question though:
**Do you confirm this is the right way to use the profiler on VART-like applications? Since examples are given for Vitis-AI-Library samples, I have a slight doubt here.**

Regards

Fabioni, 2021-06-01:
Hey @hanxue and @foolmarks, did you manage to find how to do the fine-grained profiling in the new Vitis-AI? I am struggling with that for days.

li900309, 2022-03-10:
hi,
the latest vaitrace can start a fine grained profiling in a simple way, just add --fine_grained option in the vaitrace cmd , please have a look at this doc:  https://github.com/Xilinx/Vitis-AI/tree/master/tools/Vitis-AI-Profiler#command-line-usage

closed 2022-03-10 by qianglin-xlnx