Xilinx/Vitis-AI #100: Vitis-AI 1.1 runner.so file from libvart-1.1.0-Linux-build<xx>.deb for Python3.6

jlamperez, 2020-05-09:
Hi, 

In Vitis-AI 1.0 I was using something like this [runner.py](https://github.com/Xilinx/Vitis-AI/blob/master/alveo/examples/vitis_ai_alveo_samples/inception_v1_mt_py/runner.py) to load the model using a meta.json, but now in the new release Vitis-AI 1.1 I am seeing that this is installed in _/usr/lib/python3.5/site-packages/runner.so_. 

At first in **meta.json** I was using  *libvart_dpu.so* library but now *libvart-dpu-runner.so* is used and it seems that there have been some changes between this shared libraries which makes it impossible to me to use previous *runner.py*.

My problem now is that pynq image is using python3.6 and I cannot load **runner.so**. Is there any chance of getting *runner.so* for python3.6 version?

**runner.so** seems to be created using pybin11 where can I find the source code?

Thank you and regards,
Jorge

qianglin-xlnx, 2020-05-27:
Hi @jlamperez 
Since the VART code is not open source, it's hard for you to build runner.so using python3.6. However, DPU-PYNQ has been released. You can try to use the dpu-pynq image with DPU inside.  Please refer to the following link for more details.
https://github.com/Xilinx/DPU-PYNQ

yunqu, 2020-06-19:
@jlamperez Please check the latest DPU-PYNQ release. We have VART now supported.

jlamperez, 2020-06-19:
Thank you for your support @yunqu

closed 2020-06-19 by jlamperez