Xilinx/Vitis-AI #99: Yocto recipe of your toolchain

finn-ball, 2020-05-09:
So it seems pretty clear that your toolchain is built with Yocto. Is this recipe available somewhere to download?

qianglin-xlnx, 2020-05-11:
Hi @finn-ball,
I think you can find what you need in the following link.
https://github.com/Xilinx/Vitis_Embedded_Platform_Source/tree/master/Xilinx_Official_Platforms/zcu102_dpu

qianglin-xlnx, 2020-05-28:
Hi @finn-ball 
Hope your issue has been solved. If not, find recipe in the following link. 
https://github.com/Xilinx/Vitis_Embedded_Platform_Source/tree/2019.2/Xilinx_Official_Platforms/zcu102_dpu/petalinux
The "sdk.sh" is generated by the following command.
[image]

finn-ball, 2020-05-28:
Thanks, this information is what I was looking for. Perhaps add a link to this in your README for other users?

qianglin-xlnx, 2020-05-28:
What you said can be found in TRD section especially for the user who want to customize the sysroot. For most users, we provide an off-the-shelf cross-tool chain installation package that they don't have to regenerate themselves.

https://github.com/Xilinx/Vitis-AI/tree/master/DPU-TRD/prj/Vitis
[image]

qianglin-xlnx, 2020-06-15:
Hi @finn-ball 
Since this issue has been resolved and I haven't received your feedback for a long time, I will close the issue.

closed 2020-06-15 by qianglin-xlnx