Xilinx/Vitis-AI #102: Is netune example is still work?

lllkiller-jlll, 2020-05-15:
Hi.

I'm trying to run neptune example in alveo directory.

But in my environment, the example is not work.

I started server as typing "./run.sh" and I accessed <host_name>:8998.

I was trying to start services "sface", "simnet", "scoco" but my host's console printed error below: 
```
Speaking to Butler 
Response from Butler is: 
errCode: errCode: 4
errCode String: ASSIGNMENT_UNFEASIBLE
myHandle: 0
valid: 1
```

and I also saw my xbutler's console view. The xbutler's console printed message below:
```
-------------------------------------
        New Message From Client
-------------------------------------
Server: got connection on Fri May 15 15:36:43 2020 
Server: got connection from 127.0.0.1 port 42450
ServerBooklet: Already exist socket: 7.
ServerBooklet: Already exist port: 53925.
Servicing PID 328
-------------------------------------
Starting AcquireCU...
AcquireCU: Receiving user name...
AcquireCU: user name is: root
ServerBooklet: Already exist socket: 7.
ServerBooklet: Already exist username: root.
AcquireCU: Receiving kernel name...
AcquireCU: Kernel name is: kernelSxdnn_0
AcquireCU: Receiving xclbin batch...
AcquireCU: Non-UDF Mode.
AcquireCU: executing algo.
AcquireCU: Validating UDF result...
ERROR: AcquireCU: FAILED!
AcquireCU: Sending response...
AcquireCU: Response is: ASSIGNMENT_UNFEASIBLE
AcquireCU: Sending handle...
AcquireCU: Handle is: 0
AcquireCU: Sending result...
Finishing AcquireCU.
-------------------------------------
-------------------------------------
        New Message From Client
-------------------------------------
Server: got connection on Fri May 15 15:36:44 2020 
Server: got connection from 127.0.0.1 port 42450
ServerBooklet: Already exist socket: 7.
ServerBooklet: Already exist port: 53925.
-------------------------------------
      Detected Dead Connection
Removing socket: 7
ServerBooklet: Removing socket: 7.
-------------------------------------
```

How can I start all services for this example?

please let me know.

Thanks.

hanxue, 2020-08-03:
Does restarting xbutler help?

```
xbutil reset
sudo service xbutler restart
```

You can also try reinstalling XButler, for example on Ubuntu

```
sudo apt install ./xbutler_2.0-6.deb –reinstall
```

matejbizjak, 2021-05-06:
I also have the same problem on Alveo U250 card. I'm using both Alveo U250 and U280 cards and I can successfully run examples made for U280, but I was unable to run examples for U250. 

I followed [this script](https://github.com/Xilinx/Vitis-AI/blob/master/setup/alveo/u200_u250/packages/install.sh) and installed the packages manually. Xbutler is working properly as well as the XRT and the cards. 

This is the output of the `source ${VAI_HOME}/setup/alveo/u200_u250/overlaybins/setup.sh` command:
```
------------------
Using VAI_HOME
------------------
/vitis_ai_home

---------------------
Verifying XILINX_XRM
---------------------
---------------------
Using LD_LIBRARY_PATH
---------------------
/opt/xilinx/xrt/lib:/opt/vitis_ai/conda/envs/vitis-ai-caffe/lib:/opt/xilinx/xrt/lib:/usr/lib:/usr/lib/x86_64-linux-gnu:/usr/local/lib:/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib
--------------------
Vitis-AI Flow
---------------------
-------------------
Using LIBXDNN_PATH
-------------------
/opt/vitis_ai/conda/envs/vitis-ai-caffe/lib/libxfdnn.so

-------------------
PYTHONPATH
-------------------
/opt/xilinx/xrt/python:

---------------------
Verifying XILINX_XRT
---------------------
XILINX_XRT        : /opt/xilinx/xrt
PATH              : /opt/xilinx/xrt/bin:/opt/xilinx/xrt/bin:/opt/vitis_ai/conda/envs/vitis-ai-caffe/bin:/opt/vitis_ai/conda/bin:/opt/vitis_ai/conda/condabin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
LD_LIBRARY_PATH   : /opt/xilinx/xrt/lib:/opt/vitis_ai/conda/envs/vitis-ai-caffe/lib:/opt/xilinx/xrt/lib:/opt/vitis_ai/conda/envs/vitis-ai-caffe/lib:/opt/xilinx/xrt/lib:/usr/lib:/usr/lib/x86_64-linux-gnu:/usr/local/lib:/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib
PYTHONPATH        : /opt/xilinx/xrt/python:/opt/xilinx/xrt/python:
```
But when I run the DPUCADX8G examples I get the following errors... for example:
~/build/build.Ubuntu.18.04.x86_64.Release/Vitis-AI-Library/classification/test_classification inception_v1 samples/classification/images/001.JPEG 

```
-------------------
Speaking to Butler 
Response from Butler is: 
errCode: errCode: 4
errCode String: ASSIGNMENT_UNFEASIBLE
myHandle: 0
valid: 1
-------------------
....
```
I have already read [this issue](https://github.com/Xilinx/Vitis-AI/issues/33) but it doesn't help me.