Xilinx/Vitis-AI #103: When I install docker gpu,meet one error.

zy486at189cn, 2020-05-16:
./docker_run.sh xilinx/vitis-ai-gpu:latest
system report:
docker: Error response from daemon: OCI runtime create failed: container_linux.go:345: starting container process caused "process_linux.go:430: container init caused \"process_linux.go:413: running prestart hook 0 caused \\\"error running hook: exit status 1, stdout: , stderr: nvidia-container-cli: requirement error: unsatisfied condition: cuda>=10.0\\\\n\\\"\"": unknown.
Why?
cuda should be 10.0?or other?

surekha1723, 2020-05-20:
The nvidia-driver in your server should be latest one.

hanxue, 2020-08-03:
For this error message
```
 stderr: nvidia-container-cli: requirement error: unsatisfied condition: cuda>=10.0\\n\""
```

Docker host is expecting to find NVIDIA driver that can run CUDA version 10.0. You need to install **NVIDIA driver >=410.78** to resolve this problem.

closed 2020-08-03 by hanxue